gazebosim/gz-physics #67: Create ign-physics tutorials

claireyywang, 2020-06-17:
Some tutorial ideas 

Getting started

- [ ] what is ign-physics, motivation of creating this library, where to find repo, versions released
- [ ] currently supported physics engines and corresponding features overview
- [ ] roadmap of future plans of ign-physics, potential timeline of certain features to look forward to
- [ ] where to download debian, how to build/compile 

For users

- [ ] how to switch between default and additional physics engine
- [ ] sections where users try out different physics features
  - [ ] bottom up approach: step by step tutorials on how to set/add various models with different physics features like (rigid body dynamics, collision detection, kinematics to compose a world
OR  
   - [ ] top down approach: given a world of various models, zoom in onto each model and see how forces are set, position is updated or any other physics features work

For developers

- [ ] High level overview of how physics plugin work, explain user interface and implementation interface, key concepts/classes, link to API documentation and contributing guide
- [ ] explain folder structure and where to add new custom features, potential restrictions on adding custom features 
- [ ] a step by step interactive tutorial teaching how to add a simple custom plugin, using the [hello-world plugin example](https://github.com/ignitionrobotics/ign-physics/tree/ign-physics2/examples)
- [ ] use TPE as a more advanced example. Pick one `FeatureList` as step-by-step tutorial on how to implement features using external physics engine API.

chapulina, 2020-06-17:
I'll leave a cross-reference to this issue because some questions that are popping up there could be good to answer on these tutorials: https://github.com/leggedrobotics/raisimLib/issues/40

scpeters, 2020-06-21:
I think it would also be make a tutorial about how to define custom features in a plugin and request them when loading plugins. We can use the [dartsim RetrieveWorld feature in World.hh](https://github.com/ignitionrobotics/ign-physics/blob/master/dartsim/include/ignition/physics/dartsim/World.hh#L30) as an example.

claireyywang, 2020-06-23:
> I think it would also be make a tutorial about how to define custom features in a plugin and request them when loading plugins. We can use the [dartsim RetrieveWorld feature in [...]

Good point! Adding it to the list...

claireyywang, 2020-11-16:
Closing since most PRs are merged and tracked in separate issues.

closed 2020-11-16 by claireyywang