gazebosim/gz-physics #68: Annoying GenericJoint::setRestPosition DART warning

diegoferigo, 2020-06-18:
I am running fairly long simulations (~days) and the log is flooded with warnings like the following:

```
Warning [GenericJoint.hpp:1473] [GenericJoint::setRestPosition] Value of _q0 [0], is out of the limit range [0.261799, 1.85005] for index [0] of Joint [Joint].
```

There is no way to suppress it since it's related to the DART logger.

Anyway, this warning has been removed in the upstream repository in https://github.com/dartsim/dart/pull/1418, but the DART packages in the osrf repository does not seem to include this fix. If I recall, you use the [azeey/dart](https://github.com/azeey/dart) fork for packaging, that is a bit behind upstream.

Waiting https://github.com/ignitionrobotics/ign-physics/issues/49, would it be possible to have this upstream PR merged in the next release? @azeey @j-rivero

scpeters, 2020-06-20:
that sounds reasonable to me

azeey, 2023-12-11:
Since we're now mostly using upstream DART with 6.12 and later, I think it's safe to close this.

closed 2023-12-11 by azeey