gazebosim/gz-physics #72: Fail to build ignition-physics

MakinoharaShouko, 2020-06-25:
When building ignition-dome on Ubuntu 18.04 following [this document](https://github.com/MakinoharaShouko/rmf_demos/blob/feature/add_ignition_support/docs/ignition.md), I got the following error  
I am wondering how can this be fixed
```
--- stderr: ignition-physics3                                                
CMake Error at /usr/share/dart/cmake/DARTFindurdfdom.cmake:13 (set_target_properties):
  Property INTERFACE_LINK_LIBRARIES may not contain link-type keyword
  "optimized".  The INTERFACE_LINK_LIBRARIES property may contain
  configuration-sensitive generator-expressions which may be used to specify
  per-configuration rules.
Call Stack (most recent call first):
  /usr/share/dart/cmake/dart_utils-urdfComponent.cmake:6 (include)
  /usr/share/dart/cmake/DARTConfig.cmake:63 (include)
  /usr/share/dart/cmake/DARTConfig.cmake:123 (dart_traverse_components)
  /usr/share/dart/cmake/DARTConfig.cmake:171 (dart_package_init)
  /home/makinohara/ws_dome/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:188 (find_package)
  CMakeLists.txt:63 (ign_find_package)

---
Failed   <<< ignition-physics3 [0.40s, exited with code 1]
```

traversaro, 2020-06-25:
The error seems to be an incompatibility between DART's CMake script and the `urdfdom`  installed by ROS2,  see https://github.com/ros2/urdfdom/pull/16 and https://github.com/robotology/gym-ignition/issues/118#issuecomment-575372388 . 
In theory the problem should be fixed  by https://github.com/ros2/urdfdom/pull/16  , but I do not know if the fix has  been released, from the document you linked it  should  be fixed on ROS 2 Foxy, which version of ROS2 do you have installed  instead?

MakinoharaShouko, 2020-06-25:
> The error seems to be an incompatibility between DART's CMake script and the `urdfdom` installed by ROS2, see [ros2/urdfdom#16](https://github.com/ros2/urdfdom/pull/16) and [robotology/gym-ignition#118 (comment)](https://github.com/robotology/gym-ignition/issues/118#issuecomment-575372388) .
> In theory the problem should be fixed by [ros2/urdfdom#16](https://github.com/ros2/urdfdom/pull/16) , but I do not know if the fix has been released, from the document you linked it should be fixed on ROS 2 Foxy, which version of ROS2 do you have installed instead?

I am using ROS eloquent and I found it does not contain the fix. I will manually modify that.

closed 2020-06-25 by MakinoharaShouko