gazebosim/gz-physics #82: Expose contact parameters

diegoferigo, 2020-07-15:
In our simulations with a humanoid robot we are noticing some funky contact detection behaviour. It's something similar to what was shown in https://github.com/robotology/gym-ignition-models/pull/20. Particularly, when the feet look visually flat on the ground plane, the contacts detected by DART are quite noisy, they go on and off.

In Gazebo Classic, ODE could be tuned quite a lot using the [`/collision/contact/ode`](http://sdformat.org/spec?ver=1.7&elem=collision#contact_ode) parameters. However, currently none of them are exposed for DART.

Some while ago @scpeters in https://github.com/robotology/icub-gazebo/issues/30#issuecomment-220400178 mentioned that the `min_depth` parameter is quite relevant.

Is 1) setting the depth currently possible in DART and 2) did you ever had any internal discussion about these set of parameters? I suspect that they can become fundamental for any locomotion and manipulation applications.

cc @azeey @chapulina

diegoferigo, 2020-07-15:
Other related issues and pr: https://github.com/ignitionrobotics/ign-physics/pull/40, https://github.com/ignitionrobotics/ign-gazebo/issues/112, https://github.com/ignitionrobotics/ign-gazebo/pull/234, https://github.com/ignitionrobotics/ign-gazebo/pull/229.

scpeters, 2020-07-20:
We didn't have many contact parameters exposed in osrf/gazebo's dart layer, but it would be good to expose some here. I'm in favor of adding `ignition::physics::Feature`s that expose well-documented parameters, as long as they are optional (i.e. do not increase the required API surface area).

I would look at dart's [ShapeFrameAspect.hh](https://github.com/dartsim/dart/blob/master/dart/dynamics/detail/ShapeFrameAspect.hpp#L89-L136) header file to see what parameters can be set on a per-shape basis. From a quick glance, it looks like they support friction parameters and a restitution coefficient, but not as many parameters as were exposed in Open Dynamics Engine.

diegoferigo, 2020-11-02:
I come back to this issue after a while, I managed to dedicate some time to it because the jumpy contacts situation has become a blocker for many of our applications involving humanoid robots.

As you correctly wrote, currently DART does not expose any of the parameters that ODE has. Unfortunately, I can confirm that relaxing a bit the constraints at the physics engine level is one of the only way I found to stabilize the contacts, as I described in https://github.com/ignition-forks/dart/issues/14.

Adding in a clean way the possibility to tune that parameter in a per-contact basis, considering the architecture of DART, does not seem a trivial task.

traversaro, 2020-11-02:
I remember that back in time there was a proposal at the Gazebo Classic levle to support in a more flexible way parameters, i.e. without requiring each parameter to be exposed in the SDF spec. Now that the work to support arbitrary physics engine is going forward at the ign-physics level and the SDF specs support custom extensions, one thing that it could be good to have is some ignition-physics SDF extention to pass arbitrary parameters (in the form of key-value map, or simular) to the underlying physics engine.