gazebosim/gz-physics #87: Jenkins Windows CI is not installing DART

chapulina, 2020-07-23:
Windows CI is currently skipping the dartsim component:

```
-- Could NOT find DART (missing: DART_DIR)
-- Looking for DART - not found

-- -------------------------------------------

-- BUILD WARNINGS
-- 	Cannot build component [dartsim] - Missing: DART (Components: collision-ode, utils, utils-urdf, CONFIG)
-- END BUILD WARNINGS
```

We should make sure DART and its Ignition Physics bindings are installable and runnable on Windows.

scpeters, 2020-07-23:
I had been working on this for a while before, but dart wasn't building on windows at the time. I think it is in better shape now, but we also require several components, so we need to make sure those dependencies are available as well.

* add dart to ign-physics workspace: https://github.com/ignition-tooling/gazebodistro/commit/15d4b5a8d30b67cab0a43bcd9e8db31be20cf004
* install extra vcpkg dependencies: https://github.com/ignition-tooling/release-tools/commit/2cbf664e1aee8a76ccfb940649fe5760401379af

test build fails; I think our vcpkg state is out of date and the ode formula has a bitbucket hash? I believe @traversaro fixed this in https://github.com/microsoft/vcpkg/pull/11029, but maybe something is out of date on our CI machines?

* [image] https://build.osrfoundation.org/job/ign_physics-ign-2-win/41/

I'll try again without ODE...

scpeters, 2020-07-23:
need to fix this I think:

* https://github.com/dartsim/dart/blob/becbeada2a8ba94834e69ceb9f4efaa92381b756/CMakeLists.txt#L54

scpeters, 2020-07-25:
dart install location will be fixed by https://github.com/dartsim/dart/pull/1478

now cmake is failing while finding dart dependencies:

* [image] https://build.osrfoundation.org/job/ign_physics-ign-2-win/45/console

scpeters, 2020-09-18:
dart is now in vcpkg: https://github.com/dartsim/dart/issues/1005

so that should simplify matters if it builds the components that we need

scpeters, 2020-10-05:
latest test with dartsim from vcpkg: [image] https://build.osrfoundation.org/job/ign_physics-ign-2-win/58/

scpeters, 2020-10-05:
> latest test with dartsim from vcpkg: [image] https://build.osrfoundation.org/job/ign_physics-ign-2-win/58/

~~~
Error: Cannot find definition for package `dartsim`.
~~~

@j-rivero are the vcpkg definitions out of date on our CI machines? I was hoping to try the `dartsim` package

j-rivero, 2020-10-05:
> @j-rivero are the vcpkg definitions out of date on our CI machines? I was hoping to try the `dartsim` package

Yep, we are working with a "frozen" (we are not running git pull on the repo) snapshot of the vcpkg formulas. That is somehow good for stability although brings these kind of updated problems. My recommendation would be try to update vcpkg in a system out of the buildfarm (or take a node out of CI), git pull to the latest version and check to build the whole ignition suite.

scpeters, 2020-10-05:
> Yep, we are working with a "frozen" (we are not running git pull on the repo) snapshot of the vcpkg formulas. That is somehow good for stability although brings these kind of updated problems. My [...]

sounds good; I will work on this with @JShep1 and @chapulina

scpeters, 2020-10-07:
I just got my windows VM running again and installed `dartsim` with `vcpkg`, and it's closer, but `ign-physics` still fails to find DART's `collision-ode` component:

~~~
-- Could NOT find DART (missing: collision-ode) (Required is at least version "6.9")
~~~

more verbose log at https://gist.github.com/scpeters/d7a66c27f6423c9c8d13156f270d7eab

I'll try disabling the ode component for now and see if I can get farther

scpeters, 2020-10-07:
I've temporarily substituted the `collision-bullet` component for `collision-ode` and have started making some changes in a test branch:

* 425b82746ce38f85d84d6c18f820c9429e87122f: use `/permissive-` for dartsim-plugin per recommendation in https://github.com/dartsim/dart/issues/753#issuecomment-393067157
* 03744a814d6cd477d109eaa0f869ca58a58787d8: copy instead of symlink the unversioned plugin on windows (copied from approach in #32 for tpe)

it now has Assimp and FCL linking errors (full console output in [gist](https://gist.github.com/scpeters/d43d6c792ad0a467be89dafe4ac607c7))

~~~
CustomMeshShape.obj : error LNK2001: unresolved external symbol "public: __cdecl aiNode::aiNode(void)" (??0aiNode@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
EntityManagementFeatures.obj : error LNK2001: unresolved external symbol "public: static class std::shared_ptr<class dart::collision::OdeCollisionDetector> __cdecl dart::collision::OdeCollisionDetector::create(void)" (?create@OdeCollisionDetector@collision@dart@@SA?AV?$shared_ptr@VOdeCollisionDetector@collision@dart@@@std@@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol "public: __cdecl aiNode::~aiNode(void)" (??1aiNode@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiImportFileExWithProperties [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiApplyPostProcessing [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiReleaseImport [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiCreatePropertyStore [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiReleasePropertyStore [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiSetImportPropertyInteger [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: static void * __cdecl Assimp::Intern::AllocateFromAssimpHeap::operator new(unsigned __int64)" (??2AllocateFromAssimpHeap@Intern@Assimp@@SAPEAX_K@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl Assimp::Intern::AllocateFromAssimpHeap::operator delete(void *)" (??3AllocateFromAssimpHeap@Intern@Assimp@@SAXPEAX@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: virtual bool __cdecl Assimp::IOSystem::ComparePaths(char const *,char const *)const " (?ComparePaths@IOSystem@Assimp@@UEBA_NPEBD0@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: __cdecl fcl::Triangle::Triangle(void)" (??0Triangle@fcl@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: void __cdecl fcl::Triangle::set(unsigned __int64,unsigned __int64,unsigned __int64)" (?set@Triangle@fcl@@QEAAX_K00@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: unsigned __int64 __cdecl fcl::Triangle::operator[](int)const " (??ATriangle@fcl@@QEBA_KH@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(FCLCollisionObject.obj) : error LNK2001: unresolved external symbol "public: unsigned __int64 __cdecl fcl::Triangle::operator[](int)const " (??ATriangle@fcl@@QEBA_KH@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "void __cdecl fcl::eigen_old<double>(class Eigen::Matrix<double,3,3,0,3,3> const &,class Eigen::Matrix<double,3,1,0,3,1> &,class Eigen::Matrix<double,3,3,0,3,3> &)" (??$eigen_old@N@fcl@@YAXAEBV?$Matrix@N$02$02$0A@$02$02@Eigen@@AEAV?$Matrix@N$02$00$0A@$02$00@2@AEAV12@@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
~~~

scpeters, 2020-10-09:
> I've temporarily substituted the `collision-bullet` component for `collision-ode` and have started making some changes in a test branch:
> 
> * [scpeters@425b827](https://github.com/scpeters/ign-physics/commit/425b82746ce38f85d84d6c18f820c9429e87122f): use `/permissive-` for dartsim-plugin per recommendation in [dartsim/dart#753 (comment)](https://github.com/dartsim/dart/issues/753#issuecomment-393067157)
> * [scpeters@03744a8](https://github.com/scpeters/ign-physics/commit/03744a814d6cd477d109eaa0f869ca58a58787d8): copy instead of symlink the unversioned plugin on windows (copied from approach in #32 for tpe)
> 
> it now has Assimp and FCL linking errors (full console output in [gist](https://gist.github.com/scpeters/d43d6c792ad0a467be89dafe4ac607c7))
> 
> ```
> CustomMeshShape.obj : error LNK2001: unresolved external symbol "public: __cdecl aiNode::aiNode(void)" (??0aiNode@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> EntityManagementFeatures.obj : error LNK2001: unresolved external symbol "public: static class std::shared_ptr<class dart::collision::OdeCollisionDetector> __cdecl dart::collision::OdeCollisionDetector::create(void)" (?create@OdeCollisionDetector@collision@dart@@SA?AV?$shared_ptr@VOdeCollisionDetector@collision@dart@@@std@@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol "public: __cdecl aiNode::~aiNode(void)" (??1aiNode@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiImportFileExWithProperties [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiApplyPostProcessing [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiReleaseImport [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiCreatePropertyStore [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiReleasePropertyStore [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(MeshShape.obj) : error LNK2001: unresolved external symbol aiSetImportPropertyInteger [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: static void * __cdecl Assimp::Intern::AllocateFromAssimpHeap::operator new(unsigned __int64)" (??2AllocateFromAssimpHeap@Intern@Assimp@@SAPEAX_K@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: static void __cdecl Assimp::Intern::AllocateFromAssimpHeap::operator delete(void *)" (??3AllocateFromAssimpHeap@Intern@Assimp@@SAXPEAX@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(AssimpInputResourceAdaptor.obj) : error LNK2001: unresolved external symbol "public: virtual bool __cdecl Assimp::IOSystem::ComparePaths(char const *,char const *)const " (?ComparePaths@IOSystem@Assimp@@UEBA_NPEBD0@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: __cdecl fcl::Triangle::Triangle(void)" (??0Triangle@fcl@@QEAA@XZ) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: void __cdecl fcl::Triangle::set(unsigned __int64,unsigned __int64,unsigned __int64)" (?set@Triangle@fcl@@QEAAX_K00@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "public: unsigned __int64 __cdecl fcl::Triangle::operator[](int)const " (??ATriangle@fcl@@QEBA_KH@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(FCLCollisionObject.obj) : error LNK2001: unresolved external symbol "public: unsigned __int64 __cdecl fcl::Triangle::operator[](int)const " (??ATriangle@fcl@@QEBA_KH@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> dart.lib(FCLCollisionDetector.obj) : error LNK2001: unresolved external symbol "void __cdecl fcl::eigen_old<double>(class Eigen::Matrix<double,3,3,0,3,3> const &,class Eigen::Matrix<double,3,1,0,3,1> &,class Eigen::Matrix<double,3,3,0,3,3> &)" (??$eigen_old@N@fcl@@YAXAEBV?$Matrix@N$02$02$0A@$02$02@Eigen@@AEAV?$Matrix@N$02$00$0A@$02$00@2@AEAV12@@Z) [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> ```

@traversaro have you seen these linking errors before?

traversaro, 2020-10-10:
> @traversaro have you seen these linking errors before?

I did not, but it seems that being dart a static library on Windows (see https://github.com/dartsim/dart/issues/1501), it is necessary to explicitly link all the shared libraries on which it depends (in theory CMake should handle this), while from the errors it seems that fcl and assimp are not properly linked. For debugging, it would be helpful to have the full linker invocation line to check if fcl and assimp are actually linked.

traversaro, 2020-10-11:
By the way, the best way for me to catch this problems has always to add vcpkg ports for ignition libraries, to ensure that even if there are regression at the vcpkg level they are catched fast by vcpkg CI and dev themselves. Unfortunately in this case the ign-physics port (that should be relatively simple) has been blocked by the availability of the ignition-common port (see https://github.com/microsoft/vcpkg/pull/11273). Unfortunately the ignition-common port was quite unlucky, as at some point in the vcpkg CI they installed pkg-config as well, but without making sure before that all the ports installed valid `.pc` files. For this reason, the ignition-common ports has been waiting for a long time for the .pc files of its dependency to be fixed. In theory it is possible to modify ignition-common and ignition-cmake to optionally avoid to use the .pc files, but to be honest I have never invested time in such a complicated workaround that would not be useful in the long term.

traversaro, 2020-10-12:
> @traversaro have you seen these linking errors before?

I think I noticed something related: DART on MSVC hardcodes the library names of fcl and assimp (https://github.com/dartsim/dart/blob/0be8c3de481f85e440579ab1c60a8242a0d936c1/cmake/Findfcl.cmake#L31 and https://github.com/dartsim/dart/blob/e18046107679ff71a0da75dbbf2982b5650ff490/cmake/Findassimp.cmake#L29) instead of using the imported targets, so this could explain the linking errors. Unfortunately testing Dart on Windows is quite complicated due to the impossibility of compiling DART as a shared library (see https://github.com/dartsim/dart/issues/1501), so even just compiling the test means that you get a 1 Gb executable for each test.

traversaro, 2020-10-12:
> Unfortunately testing Dart on Windows is quite complicated due to the impossibility of compiling DART as a shared library (see [dartsim/dart#1501](https://github.com/dartsim/dart/issues/1501)), so even just compiling the test means that you get a 1 Gb executable for each test.

Actually the tests on Windows were disabled in https://github.com/dartsim/dart/pull/1286 .

scpeters, 2020-10-12:
> > @traversaro have you seen these linking errors before?
> 
> I did not, but it seems that being dart a static library on Windows (see [dartsim/dart#1501](https://github.com/dartsim/dart/issues/1501)), it is necessary to explicitly link all the shared libraries on which it depends (in theory CMake should handle this), while from the errors it seems that fcl and assimp are not properly linked. For debugging, it would be helpful to have the full linker invocation line to check if fcl and assimp are actually linked.

it took me a minute to figure out how to expose the linker command, but setting an environment variable `VERBOSE=1` did the trick. Yeah, it looks like fcl and assimp are not listed at all in the link command (with some massaged whitespace for ease of reading):

~~~
Link:
  C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\VC\Tools\MSVC\14.27.29110\bin\HostX64\x64\link.exe
  /ERRORREPORT:QUEUE
  /OUT:"C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\bin\Release\ignition-physics3-dartsim-plugin.dll"
  /INCREMENTAL:NO
  /NOLOGO
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\dart-collision-bullet.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\BulletDynamics.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\BulletCollision.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\LinearMath.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\BulletSoftBody.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\dart-utils-urdf.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\dart-utils.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\urdfdom_sensor.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\urdfdom_model_state.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\urdfdom_model.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\urdfdom_world.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\console_bridge.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\dart.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\dart-external-odelcpsolver.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\ccd.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\boost_system-vc140-mt.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\boost_filesystem-vc140-mt.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\octomap.lib"
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\octomath.lib"
  C:\Users\scpet\Desktop\ws\ignition\install\sdformat10\lib\sdformat10.lib
  "C:\Users\scpet\Desktop\vcpkg\installed\x64-windows\lib\tinyxml2.lib"
  "..\lib\Release\ignition-physics3.lib"
  "C:\Users\scpet\Desktop\ws\ignition\install\ignition-plugin1\lib\ignition-plugin1.lib"
  "C:\Users\scpet\Desktop\ws\ignition\install\ignition-common3\lib\ignition-common3-graphics.lib"
  "C:\Users\scpet\Desktop\ws\ignition\install\ignition-math6\lib\ignition-math6.lib"
  "C:\Users\scpet\Desktop\ws\ignition\install\ignition-common3\lib\ignition-common3-profiler.lib"
  "C:\Users\scpet\Desktop\ws\ignition\install\ignition-common3\lib\ignition-common3.lib"
  kernel32.lib
  user32.lib
  gdi32.lib
  winspool.lib
  shell32.lib
  ole32.lib
  oleaut32.lib
  uuid.lib
  comdlg32.lib
  advapi32.lib
  /MANIFEST
  /MANIFESTUAC:"level='asInvoker' uiAccess='false'"
  /manifest:embed
  /PDB:"C:/Users/scpet/Desktop/ws/ignition/build/ignition-physics3/bin/Release/ignition-physics3-dartsim-plugin.pdb"
  /SUBSYSTEM:CONSOLE
  /TLBID:1
  /DYNAMICBASE
  /NXCOMPAT
  /IMPLIB:"C:/Users/scpet/Desktop/ws/ignition/build/ignition-physics3/lib/Release/ignition-physics3-dartsim-plugin.lib"
  /MACHINE:X64
  /machine:x64
  /DLL "ignition-physics4-dartsim-plugin.dir\Release\CustomFeatures.obj"

~~~

traversaro, 2020-10-12:
> Actually the tests on Windows were disabled in https://github.com/dartsim/dart/pull/1286 .

Indeed, I enabled the DART tests and I have the same linking error on DART tests.

scpeters, 2020-10-12:
https://github.com/dartsim/dart/issues/1476 seems related, but they mention a different set of  missing symbols

scpeters, 2020-10-14:
nice work, https://github.com/dartsim/dart/pull/1510 fixes the FCL linking errors and most of the Assimp linking errors; I now just see the following error:

~~~
LINK : fatal error LNK1181: cannot open input file 'utf8cpp.lib' [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
~~~

I believe the `utf8cpp` linkage is coming from assimp: https://github.com/microsoft/vcpkg/blob/cebe05184119cfabcc8f5c0e71c8673c22a1487b/ports/assimp/irrlicht.patch#L10-L11

scpeters, 2020-10-14:
> nice work, [dartsim/dart#1510](https://github.com/dartsim/dart/pull/1510) fixes the FCL linking errors and most of the Assimp linking errors; I now just see the following error:
> 
> ```
> LINK : fatal error LNK1181: cannot open input file 'utf8cpp.lib' [C:\Users\scpet\Desktop\ws\ignition\build\ignition-physics3\dartsim\ignition-physics3-dartsim-plugin.vcxproj]
> ```
> 
> I believe the `utf8cpp` linkage is coming from assimp: https://github.com/microsoft/vcpkg/blob/cebe05184119cfabcc8f5c0e71c8673c22a1487b/ports/assimp/irrlicht.patch#L10-L11

full linker command at https://gist.github.com/scpeters/7f30e97696c95093752c9e7fc45f25cc

traversaro, 2020-10-14:
> nice work, [dartsim/dart#1510](https://github.com/dartsim/dart/pull/1510) fixes the FCL linking errors and most of the Assimp linking errors; I now just see the following error:
>
> ```
> LINK : fatal [...]

Cool, this comes from a PR merged 20 hours ago ( https://github.com/microsoft/vcpkg/pull/13521 ), really a fresh regression. Let's open a vcpkg issue on that: https://github.com/microsoft/vcpkg/issues/14045 .

scpeters, 2020-10-14:
wow, thanks for deciphering that and submitting the issue; I'll watch it and try again if it gets resolved

scpeters, 2020-10-15:
ok, the `utf8cpp` linkage error is fixed by https://github.com/microsoft/vcpkg/pull/14061; I'll see how far I can get now...

scpeters, 2020-10-16:
I had some trouble linking with ODE, so I switched to the `collision-bullet` component for testing and got tests compiled and running, though any tests involving the `dartsim-plugin` fail with:

~~~
unknown file: error: C++ exception with description "Access violation - no RTTI data!" thrown in the test body.
~~~

I then tried to fix the ODE linking issue and am preparing to make a pull request for that.

traversaro, 2020-10-16:
A possible debug direction may be to enable the example and the testsuite compilation and execution of DART itself, this may highlight if there are some problems in DART itself.

scpeters, 2020-10-19:
> A possible debug direction may be to enable the example and the testsuite compilation and execution of DART itself, this may highlight if there are some problems in DART itself.

yes, I've reproduced the failures in DART itself: https://github.com/dartsim/dart/issues/1522

scpeters, 2020-12-11:
it looks like https://github.com/dartsim/dart/pull/1528 may eliminate our need for the `/permissive-` workaround

donnyward, 2020-12-11:
@scpeters it is only a partial fix. There are other errors that occur without `/permissive-` that I haven't been able to sort out yet.

chapulina, 2021-01-12:
Removing this from TODO so we can focus on the rest of the stack. Until DART is fixed, `ign-physics` is supported on Windows without the dartsim component. We welcome contributions from the community.