gazebosim/gz-physics #96: Allow runtime changes to joint limits

peci1, 2020-08-17:
I'd like to change `<limit><velocity>` and `<limit><effort>` dynamically during the simulation. Currently, it seems that these values are only read when parsing the model.

I've tried this, but changing the component data has no effect:

```c++
auto axisComponent = _ecm.Component<components::JointAxis>(this->joint);
axisComponent->Data().SetEffort(2.0);
```

I also tried re-creating the component, but that also doesn't change the physics engine behavior:

```c++
auto axisComponent = _ecm.Component<components::JointAxis>(this->joint);
axisComponent->Data().SetEffort(torque);
components::JointAxis axis(axisComponent->Data());
*axisComponent = axis;
```

scpeters, 2020-08-20:
Your code snippets are from ign-gazebo, but this is an issue with ign-physics because there is no `Feature` that exposes joint limits via the API.

scpeters, 2020-08-20:
I think a `Feature` API could be added to [Joint.hh](https://github.com/ignitionrobotics/ign-physics/blob/master/include/ignition/physics/Joint.hh)

scpeters, 2022-11-15:
feature was added in #260

PR #260 (2021-07-22): Added DART feature for setting joint limits dynamically.
# 🎉 New feature

Closes #96 

## Summary
Added features `SetJointPositionLimitsCommandFeature`, `SetJointVelocityLimitsCommandFeature`, `SetJointEffortLimitsCommandFeature` which allow runtime changes to joint limits.

The only thing that doesn't work as expected is setting a position limit and controlling the joint via velocity commands - the joint will slow down when violating the limit, but not stop. This is a reported DART issue: https://github.com/dartsim/dart/issues/1583 . I at least tried to detect the situations when the bug affe [... cut]
closed 2022-11-15 by scpeters