gazebosim/gz-physics #114: ign-physics usage tutorials (user side)

anindex, 2020-10-01:
The planning for `ign-physics` usage consists of these tutorials:
- A typical "hello-world" physics engines loading (we could use existing [example here](https://github.com/ignitionrobotics/ign-physics/blob/master/examples/hello_world_loader/hello_world_loader.cc))
  - Describe how to load custom physics engines as plugins (e.g. DART, TPE, etc.)
  - Explain in general the most important functions call for loading plugins.
- A comprehensive tutorial introducing physics simulation concepts, e.g. model, link, joint up to multibody dynamics, soft body, etc.
  - They could be described using the top-down approach: using an SDF robot model.
- A guide for how to try out the most important physics features.
  - Personally, I prefer top-down approach as new users also have a top-down view, so that they could follow naturally :)
  - Given an SDF robot model, users can zoom in onto joints, links to see how the forces, poses are set.

claireyywang, 2020-10-02:
> A typical "hello-world" physics engines loading (we could use existing example here)

I think it'd be nice to have a step-by-step tutorial walking through each function block/call of the existing example (something similar to this https://index.ros.org/doc/ros2/Tutorials/Writing-A-Simple-Cpp-Publisher-And-Subscriber/), so that user can follow alone and try it out on their own machine.

claireyywang, 2020-10-16:
Current PRs 

* https://github.com/ignitionrobotics/ign-physics/pull/115
* https://github.com/ignitionrobotics/ign-physics/pull/120