gazebosim/gz-physics #126: ign-physics physics plugins tutorials (developer side)

anindex, 2020-10-10:
Some tutorials for introducing new developers to ign-physics:
- Tutorial 7: "Implement a physics plugins"
  - Guide developer how to develop a simple plugin that implements a 3D policy and includes a sample list of `Features`.
  - Link to the tutorial "Loading a physics plugins" and "Understanding a physics plugins"
- Tutorial 8: "Implement a custom feature"
  - Guide developer how to define a simple custom feature in a plugin, e.g. [RetrieveWorld](https://github.com/ignitionrobotics/ign-physics/blob/master/dartsim/include/ignition/physics/dartsim/World.hh#L30)
  - Explain the folder structure and specify where developers add new custom feature.
  - Describe potential restrictions, limitation of adding a custom feature
- Tutorial 9: "Implement features using external physics engine API"
  - Use TPE as an example, pick a FeatureList to explain step by step how to implement features based on TPE API. 

After all the tutorials are merged, there maybe another PR to organize the order, content of the tutorials to ensure the correctness and comprehension.

claireyywang, 2020-10-16:
Current PRs

* https://github.com/ignitionrobotics/ign-physics/pull/128

claireyywang, 2020-11-16:
Closing since all PRs are merged

closed 2020-11-16 by claireyywang