gazebosim/gz-physics #153: Alternative physics engine: Nvidia PhysX

diegoferigo, 2020-11-01:
With this issue I want to collect some resources and assess whether there's any interest from the community to provide support to [Nvidia Physx](https://github.com/NVIDIAGameWorks/PhysX).

Historically, Gazebo Classic supported ODE, bullet, DART, and simbody. In these past weeks, I noticed some new movement in the custom branches for implementing bullet, and this is great. For what concerns ODE and simbody, there's already a ticket open https://github.com/ignitionrobotics/ign-physics/issues/63.

Nvidia Physx [has been open-sourced](https://blogs.nvidia.com/blog/2018/12/03/physx-high-fidelity-open-source/) "recently". It is Nvidia's choice for their Isaac Sim simulator and I expect a bright future for this engine. One of the notable features is the possibility to simulate physics on GPUs, allowing to scale a simulation to hundred or thousands of models. 

I'd be super excited to see anything similar running in Gazebo. I've just found rumors about a new major release https://github.com/NVIDIAGameWorks/PhysX/issues/309#issuecomment-665751081.

mabelzhang, 2021-01-26:
That would be exciting. Just wearing my personal hat, not representing any organization (where's my GitHub account number 2). PhysX's deformable object demo is really cool. I've seen limited robotics applications using it. Do you have a short list of robotics applications you've seen (that isn't NVIDIA's own promo)? I'm wondering about the physics fidelity. Issac was something I kept an look out for a few years ago.

traversaro, 2021-01-26:
>  I've just found rumors about a new major release [NVIDIAGameWorks/PhysX#309 (comment)](https://github.com/NVIDIAGameWorks/PhysX/issues/309#issuecomment-665751081).

FYI there has been a follow up on this: https://github.com/NVIDIAGameWorks/PhysX/issues/371#issuecomment-756794998 . 
TL;DR: Unfortunately the focus of the PhysX team switched from the open source C++ SDK for integration with other tools to the use of PhysX in [Nvidia Omniverse](https://www.nvidia.com/en-us/design-visualization/omniverse/) so while they are working on it, it is not a priority and they don't have a release scheduled yet.

diegoferigo, 2021-01-26:
I missed that follow up, good catch @traversaro. It's a shame the SDK got delayed. As commented in the original issue, not much system integration seems possible from Omniverse.

@mabelzhang not really, I was impressed as well from their results, but then I couldn't find many third-party projects using PhysX. Here below few links I found in my starred projects:

- https://github.com/petrikvladimir/pyphysx
- https://github.com/Unity-Technologies/marathon-envs
- https://github.com/YunzhuLi/PyFleX

A while ago I tried to have a look to PhysX, but I found it difficult to import the SDK in an external project, see https://github.com/NVIDIAGameWorks/PhysX/issues/208 and https://github.com/NVIDIAGameWorks/PhysX/issues/249. There is an external contribution that didn't get much interests from upstream https://github.com/NVIDIAGameWorks/PhysX/pull/222.

Up to today, I think that [vcpkg/ports/physx](https://github.com/microsoft/vcpkg/tree/master/ports/physx) is a reasonable solution to install PhysX if anyone wants to have a look. As reported in https://github.com/microsoft/vcpkg/pull/13849, there no GPU support yet, but it can be a good starting point.

uahic, 2021-02-20:
There is interest from my side and the research institute (robotics) I am working for, too. We are currently working (mostly) with the good old Gazebo (minus ignition). Switching to Nvidia Isaac Sim just because of PhysX is not really what we want at this point, eventho I found it pretty nice to work with its at a very early stage and makes you kindof dependent on the Nividia eco-system. So pushing forward Ignition along with PhysX is definitly something we would appreciate and I could offer to participate in the integration

mabelzhang, 2021-02-22:
On the PhysX side, it sounds like the starting point is the vcpkg link above. I haven't tried to install it.
On the Ignition side, there's a recent effort to integrate the Bullet engine, which might give some good pointers to what's needed to hook things up https://github.com/ignitionrobotics/ign-physics/pull/208

rishikeshrmadan, 2021-04-02:
I too would like to contribute in some manner, is anyone working on this at present? In the opinion of you all, will 5.0 be easy to port to once it releases? and since the 5.0 release timeline is uncertain, if one has to work on this, going forward with 4.1 integration in the meanwhile seems to be a sound choice right?

chapulina, 2021-04-05:
> is anyone working on this at present?

Open Robotics currently has no plans to work on this, but we welcome pull requests. The plugin could also be implemented in a separate repository and does not need to be included into this one.

> will 5.0 be easy to port to once it releases? and since the 5.0 release timeline is uncertain, if one has to work on this, going forward with 4.1 integration in the meanwhile seems to be a sound choice right?

I assume you mean Ignition Physics 5.0? We just released 4.0 with [Ignition Edifice](https://ignitionrobotics.org/docs/edifice). At the moment, we can't say for sure whether 5.0 will be included into Ignition Fortress, [coming up September 2021](https://ignitionrobotics.org/docs/all/releases), or if that's going to use 4.x. Whether we bump the version or not will depend on whether breaking changes are needed until then, and we usually make this decision in the middle of the development cycle.

In any case, adding support to a new physics engine should be possible against any existing stable release. I would suggest you target Ignition Physics 2.X, which is the version used in the current LTS, [Citadel](https://ignitionrobotics.org/docs/citadel).

mabelzhang, 2021-04-05:
Might be talking about NVIDIA PhysX 5.0, which was supposed to be released in 2020 but still isn't :sweat_smile: 
https://github.com/NVIDIAGameWorks/PhysX/issues/309#issuecomment-665751081

davidljung, 2025-01-16:
Not sure if still true (and this is an old issue), but previously the Open Source PhysX code ran on the CPU and to leverage CUDA/GPU required a separate binary dll file (close source).

azeey, 2025-01-16:
Interesting, I didn't know that. Looking at https://nvidia-omniverse.github.io/PhysX/physx/5.5.0/, it says:

> ... all CPU source code is available under the simple BSD3 open source license, and NVIDIA GPU binaries are included at no cost.

which supports your point @davidljung.