gazebosim/gz-physics #156: Heightmap support for DART

chapulina, 2020-11-05:
DART supports heightmap geometries. We should tie that into Ignition Physics.

This is needed by https://github.com/ignitionrobotics/ign-gazebo/issues/237.

It can make use of this PR adding heightmaps to SDF: https://github.com/osrf/sdformat/pull/388

See [gazebo::physics::DARTHeightmapShape](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/physics/dart/DARTHeightmapShape.cc) as reference.

scpeters, 2021-01-26:
I've started work on this in the following branch, but I haven't finished the `CustomHeightmapShape` constructor and probably need to change the parameters in the `AttachHeightmapShape` API (I had initially copied the mesh API)

* https://github.com/ignitionrobotics/ign-physics/compare/ign-physics2...scpeters:heightmap2

scpeters, 2021-01-26:
still need to port the code for creating the heightmap shape:

* https://github.com/osrf/gazebo/blob/gazebo11_11.3.0/gazebo/physics/dart/DARTHeightmapShapePrivate.hh#L73-L82
* https://github.com/osrf/gazebo/blob/gazebo11_11.3.0/gazebo/physics/dart/DARTHeightmapShape.cc#L56-L68

chapulina, 2021-02-26:
Rebased @scpeters 's branch onto `main` and fixed compilation in https://github.com/ignitionrobotics/ign-physics/compare/main...chapulina/4/heightmap

PR #218 (2021-07-08): Heightmaps 🏔️
# 🎉 New feature

Closes #156

## Summary

Add an attach heightmap feature and its implementation for DART.

The current implementation has a nice feature: there are hidden holes where objects can go through :hole: :slightly_smiling_face: It may be related to https://github.com/osrf/gazebo/issues/2838

[image]

## Test it

## Checklist
- [x] Signed all commits for DCO
- [x] Added tests
- [x] Added example and/or tutorial: on `ign-gazebo`
- [x] Updated documentation (as needed)
- [ ] ~~Updated migration guide (as needed)~~
- [x] `codecheck` passed (See [contributing](https://igniti [... cut]
closed 2021-07-08 by chapulina