gazebosim/gz-physics #166: Collision check without contact constraint

joxoby, 2020-11-18:
I'm trying to build a "volume observer" plugin for `ign-gazebo`. This plugin should detect if and what models are inside a user-specified volume. The straight-forward approach seems to be to use the collision detector from the physics engine, but without adding a contact constraint when the collision happens, that is, the volume should not interfere with the overall dynamics (ghost volume).

`DART` seems to have an option for this by setting to `false` the flag [enableContact](https://github.com/dartsim/dart/blob/febad10ed34c8f341083d8625559d63c0d64d208/dart/collision/CollisionOption.hpp#L51).

From the SDF side, the element [max_contacts](http://sdformat.org/spec?ver=1.7&elem=collision#collision_max_contacts) set to `0` might be a good way for specifying this behavior.

However, as far as I could tell, this feature is not yet implemented overally.

I'm opening this issue to see if the Ignition community is interested in this feature, and if so, to gather suggestions on how to implement it.

chapulina, 2020-11-18:
This may not fit your use case exactly, but have you looked into the [PerformerDetector](https://ignitionrobotics.org/api/gazebo/4.0/classignition_1_1gazebo_1_1systems_1_1PerformerDetector.html) system?

joxoby, 2020-11-18:
> This may not fit your use case exactly, but have you looked into the PerformerDetector system?

I wasn't aware of it. Anyways, as you say, it seems that that system is for a different use case.

chapulina, 2020-11-19:
> collision detector from the physics engine, but without adding a contact constraint when the collision happens

That's what the [collide_without_contact](http://sdformat.org/spec?ver=1.7&elem=collision#contact_collide_without_contact) element in SDF is for. That works on Gazebo classic (at least with ODE, I haven't checked DART), but hasn't been ported to Ignition yet.

> see if the Ignition community is interested in this feature, and if so, to gather suggestions on how to implement it.

Adding support for `collide_without_contact` is definitely desirable :+1: I believe the first step would be parsing it on libSDFormat's [Contact](https://github.com/osrf/sdformat/blob/5a910816ef57ebd34aa4cf0c0a233750d8af88b3/include/sdf/Surface.hh#L35) class, then creating a component for it on `ign-gazebo`, and finally adding a feature to `ign-physics` which sets the flag that you found.

joxoby, 2020-11-19:
Sounds good! I'll see what I can do...