gazebosim/gz-physics #176: Build error when compiling Identity.cc

acxz, 2020-12-10:
```
[  4%] Building CXX object src/CMakeFiles/ignition-physics3.dir/Identity.cc.o
In file included from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:303,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/src/Identity.cc:18:
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/detail/Entity.hh:363:5: error: no declaration matches ‘typename FeatureT::Implementation<Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface()’
  363 |     Entity<Policy, Features>::Interface()
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/src/Identity.cc:18:
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:283:58: note: candidates are: ‘template<class PolicyT, class FeaturesT> template<class F> const typename F::Implementation<ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface() const’
  283 |       const typename F::template Implementation<Policy> *Interface() const;
      |                                                          ^~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:276:59: note:                 ‘template<class PolicyT, class FeaturesT> template<class FeatureT> typename FeatureT::Implementation<ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface()’
  276 |       typename FeatureT::template Implementation<Policy> *Interface();
      |                                                           ^~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:216:11: note: ‘class ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >’ defined here
  216 |     class Entity
      |           ^~~~~~
In file included from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:303,
                 from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/src/Identity.cc:18:
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/detail/Entity.hh:373:5: error: no declaration matches ‘const typename FeatureT::Implementation<Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface() const’
  373 |     Entity<Policy, Features>::Interface() const
      |     ^~~~~~~~~~~~~~~~~~~~~~~~
In file included from /home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/src/Identity.cc:18:
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:283:58: note: candidates are: ‘template<class PolicyT, class FeaturesT> template<class F> const typename F::Implementation<ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface() const’
  283 |       const typename F::template Implementation<Policy> *Interface() const;
      |                                                          ^~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:276:59: note:                 ‘template<class PolicyT, class FeaturesT> template<class FeatureT> typename FeatureT::Implementation<ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Policy>* ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >::Interface()’
  276 |       typename FeatureT::template Implementation<Policy> *Interface();
      |                                                           ^~~~~~~~~
/home/acxz/vcs/git/github/acxz/pkgbuilds/ignition-physics/src/ign-physics-ignition-physics3_3.1.0/include/ignition/physics/Entity.hh:216:11: note: ‘class ignition::physics::Entity< <template-parameter-1-1>, <template-parameter-1-2> >’ defined here
  216 |     class Entity
      |           ^~~~~~
make[2]: *** [src/CMakeFiles/ignition-physics3.dir/build.make:147: src/CMakeFiles/ignition-physics3.dir/Identity.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:388: src/CMakeFiles/ignition-physics3.dir/all] Error 2
make: *** [Makefile:160: all] Error 2
```

OS: Arch Linux
gcc: 10.2.0
ign-physics: 3.1.0

noctrog, 2020-12-24:
I can reproduce the same error on my machine. However, when using Clang instead of GCC, it works. I generate the build directory with:
```bash
cmake -S . -B ./build/Linux64 -G "Unix Makefiles" -DCMAKE_BUILD_TYPE="Debug" -DCMAKE_C_COMPILER=/usr/bin/clang -DCMAKE_CXX_COMPILER=/usr/bin/clang++
```

OS: Arch Linux
GCC: 10.2.0
Clang: 11.0.0
ign-physics: ign-physics3 branch

acxz, 2020-12-24:
weird I wonder if it's a regression due to gcc: 10.2.0 then?

noctrog, 2020-12-24:
Seems to be working with GCC 9.3.0 too.

chapulina, 2020-12-28:
> I wonder if it's a regression due to gcc: 10.2.0 then?

Yeah I was able to reproduce the problem with gcc 10.2.0 on Ubuntu Focal.

I have a fix on https://github.com/ignitionrobotics/ign-physics/pull/185, but I don't know if that's "the correct" way.

acxz, 2021-01-07:
@chapulina I believe we can close this now with #185 merged in

PR #185 (2021-01-04): Fix compilation with GCC 10.2.0
Resolves #176.

I don't know why GCC 10 isn't able to associate the `Policy` alias to `PolicyT` correctly. Maybe there's a more appropriate way of fixing this while still using `Policy`, I'm not sure.

I haven't tried the fix with GCC 8 or 9, nor clang. Let's see what CI says.

While at it, I also expanded `F` to `FeatureF` to keep the pattern and make it easier to find.
closed 2021-01-07 by chapulina