gazebosim/gz-physics #198: Add Ignition Utils as dependency from v4

chapulina, 2021-01-16:
The new Ignition package has minimal dependencies and provides utility functions that could be useful to Ignition Physics, like the warning suppression macros.

https://github.com/ignitionrobotics/ign-utils/

The dependency needs to be added on CI for all 3 platforms before it can be used here.

scpeters, 2021-02-12:
Closed by #213

chapulina, 2021-02-12:
For completeness, I think it would be good to explicitly add the dependency in:

- [ ] https://github.com/ignition-release/ign-physics4-release
- [ ] https://github.com/osrf/homebrew-simulation/blob/master/Formula/ignition-physics4.rb
- [ ] https://github.com/ignition-tooling/release-tools/blob/master/jenkins-scripts/docker/ign_physics-compilation.bash

scpeters, 2021-02-12:
> * [ ]  https://github.com/ignition-release/ign-physics4-release
> * [ ]  https://github.com/osrf/homebrew-simulation/blob/master/Formula/ignition-physics4.rb

it is getting ignition-utils transitively from sdformat11, so I think it's working in several cases, but I'll update these for completeness (https://github.com/osrf/homebrew-simulation/pull/1335)

> * [ ]  https://github.com/ignition-tooling/release-tools/blob/master/jenkins-scripts/docker/ign_physics-compilation.bash

I'm not sure why we are both building these from source in these docker scripts and using the nightly debs. Shouldn't the nightly debs be enough?

scpeters, 2021-02-12:
> * [ ]   https://github.com/ignition-release/ign-physics4-release

https://github.com/ignition-release/ign-physics4-release/pull/3

chapulina, 2021-02-12:
> Shouldn't the nightly debs be enough?

Argh yeah I have it on my list to address this but I never get to make time for it. A while back we decided with @j-rivero that GitHub actions would use nightlies and Jenkins would install dependencies from source. But now that `release-tools` is using `packages.apt`, it's also picking up the nightlies. I think we should filter out our Ignition / SDFormat dependencies for `release-tools` CI. The fix itself is probably a 1 liner, but we need to make sure it doesn't break anything.

PR #213 (2021-02-12): Depend on ignition-utils1 cli component
# New feature

Closes #198 and will assist with #50.

## Summary

Add dependency on ignition-utils1 and its `cli` component. The cli component will help with implementing #50.

## Test it
See if it compiles and passes CI.

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Added example world and/or tutorial
- [ ] Updated documentation (as needed)
- [X] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test [... cut]
commit 3213fde8 (2021-02-12): Depend on ignition-utils1 cli component (#213)

It will be useful for implementing test executables
with `ign physics`. Closes #198.

* Update dependencies in migration guide
* Use ign-utils warning suppression
closed 2021-04-19 by scpeters