gazebosim/gz-physics #199: Tutorial links to line-numbers are out of date

scpeters, 2021-01-16:
Some links to lines of code in a tutorial are out of date (see https://github.com/ignitionrobotics/ign-physics/pull/191/files#r557676961). They are pointing to branches rather than specific commits, and the line numbers are no longer accurate.

Also, there is a reference to `master` in that same paragraph that should be updated to `main`.

Ambuj-Pandey, 2021-02-04:
i am new to open source and want to contribute also want to work on this my very first issue . Can you please guide me ??

ahcorde, 2021-02-08:
Hi @Ambuj-Pandey, 

My recommendations:
 - read this document about how to contibute to [Ignition Robotics](https://ignitionrobotics.org/docs/all/contributing#how-to-contribute)  
 - learn [how to contribute to an Open Source Project]( https://github.com/firstcontributions/first-contributions).

Then select an issue (like this one, tagged as "good first issue") and start contributing!