gazebosim/gz-physics #216: DART: Joint velocity limits are not applied

peci1, 2021-02-21:
## Environment
* OS Version: Ubuntu Bionic
* Source or binary build? Binary, 3.1.0-1~bionic

## Description
* Expected behavior: If I set [`<joint><axis><limit><velocity>`](http://sdformat.org/spec?ver=1.6&elem=joint#limit_velocity), the velocity limit should be applied.
* Actual behavior: It is not applied, the joint's velocity is not limited.

## Steps to reproduce
1. Create a 2-link 1-joint model, let the joint be revolute. 
2. Set a limit of 0.001 on the velocity of the joint's only axis (via SDF parameter).
3. Apply some large force on the joint.
4. The connected link moves much faster than it should.

This should be fixed by fixing https://github.com/ignition-forks/dart/issues/17 . But I just wanted to post the issue here so that other people can find it and spare them of the surprise.

chapulina, 2021-06-17:
Is this fixed in the latest DART~osrf release?

peci1, 2021-06-29:
It seems the issue is fixed. I tried in SubT simulator with CTU_CRAS_NORLAB_SPOT_SENSOR_CONFIG_1, which has position controllers for legs. If I edit the SDF, put a very low limit on velocity of joint `rear_right_hip_x` and command `ign topic -t /model/X1/joint/rear_right_hip_x/0/cmd_pos -m ignition.msgs.Double -p 'data: -1.0'`, the joint moves very slowly.

closed 2021-06-29 by peci1