gazebosim/gz-physics #219: Expose API to check what links moved

chapulina, 2021-03-03:
A downstream application using Ignition Physics, such as Ignition Gazebo, likely wants to know at every iteration what links have moved after a step.

Currently, Gazebo goes over all links and gets their latest poses with [FrameDataRelativeToWorld](https://github.com/ignitionrobotics/ign-physics/blob/db42320c967607039cd597cc3513ded9db06d993/include/ignition/physics/FrameSemantics.hh#L108-L109), because it doesn't have  a way of knowing which ones changed.

## Desired behavior

We should offer a mechanism by which Gazebo could know what moved and wouldn't need to iterate over all links at every step.

## Implementation suggestion

The way Gazebo classic achieves this with the ODE engine is by registering a [callback that ODE calls for each link that has moved](https://github.com/osrf/gazebo/blob/gazebo11/gazebo/physics/ode/ODELink.cc#L124). As a comparison, Gazebo classic doesn't have an equivalent mechanism for DART, and [iterates over all links instead](https://github.com/osrf/gazebo/blob/65753090cbf0d7b75143ac0e629cf18850dac985/gazebo/physics/dart/DARTPhysics.cc#L459-L473), which makes using DART slower than ODE.

We could take the same approach here. Ignition Physics could offer a Feature that allows registering a callback, and each physics engine may choose to provide that API. I don't know if DART offers such a function or if we'd need to add that functionality upstream. My understanding is that Bullet offers a callback API, but I did a quick search and couldn't find it, maybe @Lobotuerk knows what it is?

## Alternatives considered

Instead of a callback, maybe providing a function that returns only the links that moved could also do the trick, but it may perform worse depending on how that structure is populated and passed around.

adlarkin, 2021-03-04:
Since implementing callback functionality is more of an involved task and may not work for DART, I have proposed a temporary workaround in the meantime: https://github.com/ignitionrobotics/ign-gazebo/pull/669

adlarkin, 2021-06-21:
I believe that https://github.com/ignitionrobotics/ign-gazebo/pull/678 resolves this issue for DART and TPE. Can we close this, or does something also need to be done for Bullet?

chapulina, 2021-06-21:
I'm ok closing this, the main target of the feature was the default engine, which is currently DART.

closed 2021-06-21 by chapulina