gazebosim/gz-physics #222: Does ign-physics support DART soft body simulation via `soft_contact`?

keyan, 2021-03-09:
Users of Gazebo have reported that the [DART `soft_contact` Element](http://sdformat.org/spec?ver=1.7&elem=collision#surface_soft_contact) on surfaces doesn't seem to work, see https://github.com/osrf/gazebo/issues/2255.

I am trying to evaluate `ign-physics` to determine if this element is supported in Ignition, is it?

It seems that SDF model parsing is happening [in this file](https://github.com/ignitionrobotics/ign-physics/blob/ign-physics3/dartsim/src/SDFFeatures.cc), but I don't see anything that resembles the creation of dart `SoftBodyNode` objects happening in the (seemingly correct but apparently non-functional) [Gazebo parser](https://github.com/osrf/gazebo/blob/63d11b695134c7075161f1bdc36b91db47f27f92/gazebo/physics/dart/DARTLink.cc#L87). So I am guessing the answer is no. If so, is support for this on the Ignition roadmap?

mabelzhang, 2021-03-10:
I am not an authority on this, but I believe we have not added soft bodies to Ignition. I'm not aware of existing plans to add it.

Not sure about the Gazebo-classic front.

Are you asking about Ignition or Gazebo though? In your third paragraph, you linked to SDF parsing in ign-physics, but you also linked to the Gazebo-classic parser. ign-physics is used for Ignition, Gazebo-classic (osrf/gazebo) isn't using that.

keyan, 2021-03-10:
Hi Mabel. Thanks for your response.

I am asking about Ignition, but in the context of trying to find an alternative to Gazebo-classic because that doesn't seem to have properly working soft body support.

mabelzhang, 2021-03-10:
Not to dash hopes or anything, but looks like there's a thread here regarding Gazebo-classic https://github.com/osrf/gazebo/issues/1422

I've never tried it, so I can't say for sure what the status is.

keyan, 2021-03-10:
Yeah seems like the answer is no.

Well thanks for the help, and please consider adding support for softbody contact in Ignition if the topic comes up. I believe this is better supported in Bullet than DART, if that is every integrated with `ign-physics`.

mabelzhang, 2021-03-10:
On a related note, we have active effort to add Bullet support to `ign-physics` https://github.com/ignitionrobotics/ign-physics/pull/208

chapulina, 2021-03-10:
Reopening as a feature request. There's nothing preventing adding support for soft contacts in the future, someone just needs to put some time into it.

addy1997, 2021-05-16:
@keyan, have you tried this [variant](https://github.com/simonleonard/gazebo_soft). It is based on this [paper](https://docplayer.net/31475735-Softbody-simulation-in-gazebo-for-satellite-servicing-abstract.html).

mabelzhang, 2021-11-12:
For awareness:
After a short preliminary discussion, we think it makes sense to implement the rendering part of soft bodies first, before implementing the physics. Otherwise, without rendering, it’ll be hard to debug the physics.

Just an idea for whoever that picks up the implementation. It’s probably the route we’d take if we were to do it, though we don’t have immediate plans.