gazebosim/gz-physics #233: Should rename the term "Canonical link" in FreeGroup

azeey, 2021-03-17:
## Desired behavior
The term "canonical link" in SDFormat refers to a link to which the model frame is attached. That is, there's a coordinate frame called the model frame and it has a constant pose offset from the implicit frame defined by the canonical link. On the other hand, the term "canonical link" in the context of `FreeGroup` refers to a link in the model that has no constraints. For DART, this link would be the root of a kinematic tree and it's parent joint would be `FreeJoint`, which has 6DOF. Thus it is possible that the "canonical link" of a model per SDFormat be different from the "canonical link" of the `FreeGroup` of the model

For example:
```xml
<?xml version="1.0"?>
<sdf version="1.7">
  <model name="M1">
    <link name="L1"/>
    <link name="L2"/>
    <joint name="J1" type="fixed">
      <parent>L2</parent>
      <child>L1</child>
    </joint>
  </model>
</sdf>
```
The canonical link of `M1` per SDFormat is the link `L1`, however the canonical link of the FreeGroup of `M1` is `L2` because `L1` and `L2` are constrained by a joint and `L2` is the parent. 

This is also exacerbated by the fact that starting in [SDFormat 1.7](http://sdformat.org/tutorials?tut=pose_frame_semantics&cat=specification&#canonical-link), the canonical link can be specified by the user to be any link in the model.

Since the term is overloaded, and the Gazebo/SDFormat version of canonical link existed (maybe implicitly) before the `FreeGroup` counterpart, we should come up with a new name for `FreeGroup`'s canonical link.

## Implementation suggestion
Rename the FreeGroup "canonical link" to "unconstrained link"

scpeters, 2021-03-17:
there's only a few hours left before feature freeze, but this sounds like an API change to me

chapulina, 2021-03-18:
It sounds like a good idea to me. As @scpeters pointed out, we're close to feature freeze though. We could open a pull request starting the implementation. One thing to keep in mind is that the existing public `CanonicalLink` functions will need to be tick-tocked.

scpeters, 2021-03-29:
done in #234

chapulina, 2021-03-29:
Just a reminder that `ign-gazebo` is still using `CanonicalLink`, which issues a warning. I don't think there's a PR addressing this yet.

azeey, 2021-03-29:
https://github.com/ignitionrobotics/ign-gazebo/pull/723

PR #234 (2021-03-23): Rename CanonicalLink to RootLink
# 🦟 Bug fix

Fixes #233 

## Summary
See #233 for context.
This deprecates the end-user facing function `CanonicalLink` adds a new one called `UnconstrainedLink`. 

**Note**: While the changes in this PR won't break downstream users of ign-physics, it will break any physics plugins the implement the `FreeGroup` feature.

## Checklist
- [ ] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] `codecheck` passed (See [contributing](https://ignitionrobotics.org/docs/all/contributing#co [... cut]
closed 2021-03-29 by chapulina