gazebosim/gz-physics #258: Document friction parameters for each physics engine

chapulina, 2021-05-14:
We currently support some physics parameters from the SDF spec, as well as some extra ones with custom tags, and it's difficult for users to know what they are and which ones are supported by each physics engine.

To make things more complicated, because of https://github.com/ignitionrobotics/ign-physics/issues/49, users may observe different behaviours with the same engine and `ign-physics` version, but different Ubuntu versions. See how this block of code is dependent on the DART version:

https://github.com/ignitionrobotics/ign-physics/blob/1a98780ca5f6287a650f954ff2e14701bd131fe3/dartsim/src/SDFFeatures.cc#L812-L818

I think friction deserves its own tutorial in the tutorials folder.

Parameters to document:

* `mu`
* `mu2`
* `slip1`
* `slip2`
* `fdir1`
* `ignition:expressed_in` (see https://github.com/ignitionrobotics/ign-physics/pull/224)

---

This external project comparing friction between Gazebo classic and Ignition Gazebo is relevant: https://github.com/athackst/ignition_vs_gazebo/tree/main/friction

addy1997, 2021-08-14:
@chapulina, I want to help with this. If I understand it correctly, is there a tutorial to be written highlighting the above params?

chapulina, 2021-08-16:
Thanks, @addy1997 ! Yes, you can add a tutorial to this repository's `tutorials` folder. You'll need to dig in a bit to test each parameter inside [`<friction>`](http://sdformat.org/spec?ver=1.8&elem=collision#surface_friction) with each of the supported physics engines: DART, Bullet and TPE.