gazebosim/gz-physics #264: Collisions are ignored for nested models connected to parent models with fixed joint

shameekganguly, 2021-06-18:
## Environment
* OS: Ubuntu
* Source: Built from latest bazel branches

## Description
A common model composition paradigm used in our org is to model robot arms and end-effectors separately and connect them together with fixed joints, as such:
```
<model name = "parent_model">
    <link name="parent_link">...</link>
    <include><uri>model://child</uri> <name> child_model </name></include>
    <joint name="parent_joint">
        <parent> parent_model </parent>
        <child> parent_model </child>
    </joint>
</model>
```

However, it seems like collisions in the child model are being ignored in dartsim since the update to Ignition Edifice. I have attached a simple sdf (test_cylinders.txt) with two cylinder models that shows this issue. `cylinder1` on the left is defined with a nested model structure whereas `cylinder2` on the right is defined with a flat model structure. When simulated, `cylinder2` behaves correctly and falls on to the ground plane whereas `cylinder1` simply falls through.

## Steps to reproduce
```
blaze run -c opt ign_tools:ign -- gazebo -r -v 4 test_cylinders.txt
```

## Output
https://user-images.githubusercontent.com/2412842/122488705-e17a5e00-cf92-11eb-9257-2a1e3e251d69.mov
[test_cylinders.txt](https://github.com/ignitionrobotics/ign-physics/files/6673901/test_cylinders.txt)
[image]

azeey, 2021-06-21:
I believe this is an issue in ign-gazebo. The problem stems from the recent performance improvements we've been making to ign-gazbebo. In https://github.com/ignitionrobotics/ign-gazebo/pull/678, we iterate through only the links whose absolute poses have changed. Consider a scenario where the canonical link has a change in pose but the other links don't. What will happen is that we update the pose of the model that is associated with the canonical link, as well as the pose of the canonical link relative to the model. However we skip the links whose absolute poses haven't changed, but since the `components::Pose` component in ign-gazebo is always relative to the parent model, it looks like these have moved even though their absolute poses have not changed.

Here's a simplified version of `test_cylinders` from the issue description that demonstrates the problem. 
[test_cylinders_simplified.sdf.txt](https://github.com/ignitionrobotics/ign-physics/files/6687722/test_cylinders_simplified.sdf.txt)
Here `base_link` does not have a collision shape so it will fall indefinitely. `link0_outer` on the other hand does have a collision shape and it should fall until it hits the ground plane and then stop.

Here's what it looks like when I run it in ign-gazebo 5.0.0 and 4.9.1.

https://user-images.githubusercontent.com/206116/122786771-475c3380-d27a-11eb-96a7-364f03bad2a5.mp4

And this is in ign-gazebo 3.8.0

https://user-images.githubusercontent.com/206116/122786817-517e3200-d27a-11eb-8eb6-29d0595bec5b.mp4

/cc @adlarkin

azeey, 2021-06-21:
Looks like temporary fixing the issue I described above doesn't resolve the original issue with nested models raised by OP.

shameekganguly, 2021-06-21:
Thanks for looking into this @azeey.
I filed a related issue regarding pose updates for nested models here:
https://github.com/ignitionrobotics/ign-gazebo/issues/871
which I am guessing is caused by the optimizations you mentioned above.

azeey, 2021-06-23:
This should be partially fixed by #268.

PR #268 (2021-06-24): [dartsim] Fix collision detection issue when using joints across nested models
# 🦟 Bug fix

Fixes #264 

## Summary
When a joint is created between BodyNodes in different skeletons, the child BodyNode is moved to the skeleton of the parent BodyNode. When this happens, the BodyNode version of the child needs to be incremented. This is actually fixed by https://github.com/dartsim/dart/pull/1445, but we don't have that merged into our fork. So in the meantime, we call `incrementVersion` after `moveTo` is called similar to https://github.com/ignitionrobotics/ign-physics/pull/31.

## Checklist
- [x [... cut]
closed 2021-06-24 by azeey