gazebosim/gz-physics #273: Bullet plugin does not work in Edifice

diegoferigo, 2021-06-30:
## Environment

* OS Version: Ubuntu 20.04
* Source or binary build? Source at https://github.com/ignitionrobotics/ign-physics/commit/a20dbbab2bc4ed324fb3fddf060b674755234b27 and ign-gazebo 5.0 binaries (I suspect also the upcoming 5.1)

## Description

* Expected behavior: being able to load the bullet physics plugin in Edifice
* Actual behavior: the Physics system fails with the following error:

```
[Wrn] Plugin [ignition::physics::bullet::Plugin] misses required features:
- N8ignition7physics26RemoveNestedModelFromModelE
[Err] Failed to load a valid physics engine from [/workspace/install/lib/ign-physics-4/engine-plugins/libignition-physics-bullet-plugin.so].
```

## Additional details

1. This new physics feature was introduced in https://github.com/ignitionrobotics/ign-physics/pull/230.
1. Bullet reached the Edifice branch in https://github.com/ignitionrobotics/ign-physics/pull/271.
1. This new physics feature is part of:
   https://github.com/ignitionrobotics/ign-physics/blob/a20dbbab2bc4ed324fb3fddf060b674755234b27/include/ignition/physics/RemoveEntities.hh#L112-L115
1. The `RemoveEntities` feature is part of [`MinimumFeatureList`](https://github.com/ignitionrobotics/ign-gazebo/blob/276ec0f983f5007db11865a356e67a8d59ebc2f1/src/systems/physics/Physics.cc#L143-L153).

I think that currently bullet works in Dome, even though I didn't try it. However, in Edifice it seems that it cannot be loaded due to this missing feature.

cc @Lobotuerk @azeey

chapulina, 2021-06-30:
Oh I thought I had tried it with `ign-gazebo` when I did https://github.com/ignitionrobotics/ign-physics/commit/a1a8f8b3b1c1cf2f95c0acf35220009bbbb334b4. I guess I didn't :confused: 

We should update Gazebo's `MinimumFeatureList` to use `RemoveModelFromWorld` instead of `RemoveEntities`.

azeey, 2021-06-30:
Yeah, I think `RemoveNestedModelFromModel` should be optional since `ConstructSdfNestedModel` is optional.

closed 2021-07-23 by chapulina