gazebosim/gz-physics #278: 👩‍🌾 UNIT_LinkFeatures_TEST test regression on Ubuntu Bionic

caguero, 2021-07-16:
I failed two days in a row:

https://build.osrfoundation.org/job/ignition_physics-ci-main-bionic-amd64/31/
https://build.osrfoundation.org/job/ignition_physics-ci-main-bionic-amd64/32/

The test output is:

```
Unable to find test results for UNIT_LinkFeatures_TEST.xml, test did not run. Expected results in /home/jenkins/workspace/ignition_physics-ci-main-bionic-amd64/build/test_results/UNIT_LinkFeatures_TEST.xml
```

So maybe this is causing a segfault.

scpeters, 2021-07-16:
see discussion in https://github.com/ignitionrobotics/ign-physics/pull/261#issuecomment-878860217

to fix it, we need to merge https://github.com/ignitionrobotics/sdformat/pull/623 forward 11 -> 12

scpeters, 2021-07-19:
> see discussion in [#261 (comment)](https://github.com/ignitionrobotics/ign-physics/pull/261#issuecomment-878860217)
> 
> to fix it, we need to merge [ignitionrobotics/sdformat#623](https://github.com/ignitionrobotics/sdformat/pull/623) forward 11 -> 12

merged forward in https://github.com/ignitionrobotics/sdformat/pull/627

looks fixed to me:

[image] https://build.osrfoundation.org/job/ignition_physics-ci-main-bionic-amd64/33/

closed 2021-07-19 by scpeters