gazebosim/gz-physics #284: Multiple objects in constant collision

joxoby, 2021-08-23:
Hi all,

I have a use case in which multiple objects are in constant collision with each other. The default DART physics engine seems to struggle with this case, slowing down the sim rate (RTF) to 25% on my machine. In contrast, when using the Bullet physics engine (https://github.com/ignitionrobotics/ign-physics/pull/208), the simulation seems to run smoothly at a ~95% rate. See images.

What's interesting is that when using DART physics engine but selecting `dart::collision::BulletCollisionDetector` as the collision detector, the sim still runs slow at a 25% rate. Can anyone provide an explanation, or give some intuition regarding this phenomenon? I would much rather keep using DART with an improved collision detector than switching the physics engine.

[image]
[image]

mxgrey, 2021-09-27:
One possible explanation for the difference is that Bullet is a physics engine meant for video game development, and I believe it's common practice for such engines to skip simulating the physics for objects that have come to rest, until they get perturbed or acted upon by a moving object. If Bullet is using that simulation shortcut, that could easily explain the significant difference, since right now DART does not use any such shortcut.

chapulina, 2021-09-27:
> skip simulating the physics for objects that have come to rest, until they get perturbed or acted upon by a moving object

As a reference, on SDF that's toggled by the `allow_auto_disable` flag:

http://sdformat.org/spec?ver=1.8&elem=model#model_allow_auto_disable

That flag is not supported on Ignition Physics right now, but it's possible that this is the default behaviour for Bullet, I haven't checked.

abstanton, 2022-08-04:
Is there a fundamental reason why allow-auto-disable can't be implemented? Also struggling when dealing with a large number of entity collisions, however almost all of them should be static 95% of the time so allow_auto_disable would be a big help for us. If there is no obstacle in the way to implementing it I would be happy to have a look.