gazebosim/gz-physics #334: Windows CI linker issue on main

chapulina, 2022-03-15:
## Environment
* OS Version:  Focal
* Source or binary build? Source
  
  

## Description
* Expected behavior:  CI passes
* Actual behavior:  CI fails

## Steps to reproduce

Run Windows CI for the `main` branch

## Output

https://build.osrfoundation.org/job/ign_physics-pr-win/1644/

```
C:\Program Files (x86)\Microsoft Visual Studio\2019\Community\MSBuild\Microsoft\VC\v160\Microsoft.CppCommon.targets(997,5): error MSB6006: "link.exe" exited with code -1073741571. [C:\Jenkins\workspace\ign_physics-ci-win\ws\build\ignition-physics6\test\integration\INTEGRATION_FrameSemantics3f.vcxproj]
```

scpeters, 2022-04-01:
I just worked on reproducing this in a Windows VM. I had trouble getting the `vcpkg` dependencies to install with Visual Studio 2019, so I upgraded to Visual Studio 2022. With that version I was not able to reproduce the linking error.

Blast545, 2022-04-01:
:male_detective: FYI this is affecting all the windows physics jobs in the buildfarm are affected by this issue.

https://build.osrfoundation.org/job/ign_physics-ign-2-win/93/
https://build.osrfoundation.org/job/ign_physics-ign-5-win/12/
https://build.osrfoundation.org/job/ign_physics-ci-win/152/

Also this issue seems to come with a "time regression", most of the builds with this error are taking an extra hour to build/fail.