gazebosim/gz-physics #349: Build Error

onurtore, 2022-05-22:
## Environment
* OS Version:  Ubuntu 20.04
* Source installation, main branch (gz-physics6), commit (ce03083e47a60b381304e0698e5cccaf4937bb89)

## Description

Expected Behaviour: Succesfully build the ignition-physics.
Actual Behaviour: Building processs terminated with the output:

```
[Processing: ignition-physics6]                                          
--- stderr: ignition-physics6                                            
c++: fatal error: Killed signal terminated program cc1plus
compilation terminated.
make[2]: *** [dartsim/CMakeFiles/UNIT_EntityManagement_TEST.dir/build.make:63: dartsim/CMakeFiles/UNIT_EntityManagement_TEST.dir/src/EntityManagement_TEST.cc.o] Error 1
make[1]: *** [CMakeFiles/Makefile2:2684: dartsim/CMakeFiles/UNIT_EntityManagement_TEST.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make: *** [Makefile:163: all] Error 2
---
Failed   <<< ignition-physics6 [6min 9s, exited with code 2]
                                        
Summary: 13 packages finished [19min 0s]
  1 package failed: ignition-physics6
  3 packages had stderr output: ignition-physics6 ignition-rendering7 sdformat13
  2 packages not processed

```

Just re-running the build command fixes this issue, my computer also freezes couple of times during the building process so it might be a memory leak problem with colcon as well.

## Steps to reproduce
Following the tutorial [here](https://gazebosim.org/docs/garden/install_ubuntu_src) with  command _colcon build --merge-install_

azeey, 2022-05-26:
This package uses a lot of C++ template metaprogramming so , so you're likely running out of memory while building in parallel. Can you try building with a single processes? With colcon, you can try `colcon build --executor sequential`

Rnishu, 2024-03-28:
@onurtore ,

How many jobs did you have to limit it to? Is there a good way to figure out the optimal number to limit it to by seeing memory and number of cpu cores in a system?

Also, @chapulina I think this should be added to the source build instructions on gazebosim.org/docs as even with fairly newer hardware and 16gb of memory I had to limit the sequential execution down to 2 jobs.

onurtore, 2024-03-31:
@Rnishu 

Unfortunately, I do not remember the number of jobs, If I remember correctly this issue was related to some tests taking too much memory. I think I just set it to 1 after the memory problem. There should be an optimal way of doing it, but just limit it to 1.

Also @chapulina does not work for OSRF anymore, but I suggest you to create a PR for it so that everyone will benefit from it.