gazebosim/gz-physics #432: Support mimic constraint (dartsim)

adityapande-1995, 2022-10-04:
This issue keeps track of PRs related to mimic joint feature in dartsim.

* Proposal in ``sdformat`` : https://github.com/gazebosim/sdf_tutorials/pull/62
* Feature in ``gz-physics`` : https://github.com/gazebosim/gz-physics/pull/431
* Use feature inside physics plugin in ``gz-sim`` : TODO

jslee02, 2025-11-23:
I was able to reproduce the failures and fixed them in dartsim/dart#2247. The branch adds the same mimic pendulum example and integration tests that mirror the Gazebo world; with the changes, the mimic motors run without the ODE crash and the followers track their references. Please take a look and confirm this matches the expected behavior.

kseczhouyb, 2026-09-21:
Adding binary-level evidence from the packaged Harmonic stack (Gazebo Sim 8.12.0 / gz-physics 7.8.0 dartsim plugin / DART 6.13.2 / gz_ros2_control 1.2.19, ROS Jazzy / Ubuntu 24.04, OSRF apt noble):

- `nm -DC libgz-physics7-dartsim-plugin.so.7 | grep -ci mimic` → **0** (feature not implemented/exported in the shipped plugin)
- `nm -DC libdart.so.6.13 | grep -ci mimic` → **28** (DART core provides mimic support, `setMimicJoint` family)

So the gz-physics API declares `SetMimicConstraint` (`Joint.hh`: dof, leaderJoint, leaderAxisDof, multiplier, offset → bool) and DART 6.13 core supports mimic joints, but the dartsim engine plugin does not wire it. At runtime gz-sim logs per joint:

`Attempting to create a mimic constraint for joint [<joint>] but the chosen physics engine does not support mimic constraints`

In our underactuated hand model all 8 mimic joints end up unconstrained — the fingers silently degrade to free-swinging joints.

Note: gz_ros2_control provides a plugin-level mimic *emulation* (velocity-servo follower in `GazeboSimSystem::write`; offset handling fixed in 1.2.20 #953/#954) which covers ros2_control-managed joints — but an engine-level kinematic constraint is a different fidelity class (pure-gz models, pre-controller phases), and no engine-level option currently exists on dartsim.

Impact: any underactuated/mimic mechanism (robotic hands, linkage-driven joints) simulated with the dartsim engine on Harmonic.

scpeters, 2026-09-21:
it's true that mimic joints have not yet been enabled in the gz-physics-dartsim plugin, but I noticed some numerical instability (see https://github.com/dartsim/dart/pull/2247#pullrequestreview-3498393009), so I haven't enabled it yet. I suppose we could open a draft PR to add support so we can review how unstable it is

jslee02, 2026-10-12:
Sorry this took a while, but I finally tracked it down. The instability was in DART's `CouplerConstraint`: its LCP row measured only the follower's response and ignored the leader's velocity, so the follower lagged the leader by one step and coupled joints gained energy or diverged. It's fixed in dartsim/dart#3700 (`main`) and dartsim/dart#3702 (DART 6.20.1), which have the details and before/after results.

With a local `SetMimicConstraint` wiring for the dartsim plugin, gz-physics 8.0.0's mimic tests pass on the fixed DART, except `PendulumsFastSlowMimicTest`'s cross-model velocity comparison at 3 of 300 steps. That comes from the test world, not the mimic: the plugin's ground box top sits exactly at z = 0, where the base plates rest, so their contacts flicker, and two uncoupled copies diverge just as much. Raising the ground collision by 0.5 mm, or fixing the bases to the world, makes it pass.

kseczhouyb, 2026-10-12:
Thanks @jslee02 for tracking this down — the leader-velocity gap in the `CouplerConstraint` LCP row (follower-only velocity measurement, so the follower lags one step and coupled joints gain energy) explains exactly the numerical instability @scpeters saw when considering plugin enablement back in September, and the `PendulumsFastSlowMimicTest` artifact analysis (test-world ground box top at z=0 → contact flicker at 3/300 steps; raising it 0.5 mm or locking the base clears it) is a very useful clarification.

For context: we are on the packaged Harmonic stack (gz-physics 7.8.0 dartsim plugin + DART 6.13.2 from the OSRF apt repo on noble), where — as we reported above on Sep 21 — the dartsim plugin has no `SetMimicConstraint` wiring and our 8-joint underactuated hand degrades to free-swinging fingers. Since `CouplerConstraint` does not exist at all in DART 6.13.x (it first appears in the 6.20 line), local wiring alone is not an option for us. Six questions that would shape our consumption path:

1. **6.20.1 timing & packaging** — when is DART 6.20.1 expected to be tagged? The milestone shows 0 open issues, and #3702 landed on `release-6.20` *after* v6.20.0 was published, so a source build today has to track the branch head; a tagged 6.20.1 would give us a clean anchor. Also, any chance of 6.20 packaging reaching the OSRF apt repo (noble currently ships DART 6.13.2 and 6.16.6 only)?

2. **Upstreaming the plugin wiring** — do you intend to upstream your local dartsim `SetMimicConstraint` wiring as a (draft) PR? If so, would it target gz-physics main/9.x only, or also the gz-physics7 (Harmonic) maintenance line (branch-forward precedent exists, e.g. #431 → #517)? The `SetMimicConstraint` API declaration and the common mimic tests (`joint_mimic_features.cc` + worlds, including `PendulumsFastSlowMimicTest`) already exist in the 7.8.0 tree (CMake requires DART >= 6.10), so a DART-version-guarded wiring should be structurally identical there.

3. **Older DART 6 lines / Harmonic users** — is there any plan or appetite to backport the coupler chain to older lines such as 6.13.x used by Gazebo Harmonic? We note there is no `release-6.13` branch and the coupler feature (#3164) postdates 6.13, so we assume 6.20+ is the only supported surface for the fixed mimic constraint, and downstreams on 6.13.2 are expected to self-maintain — a confirmation would help us commit to a 6.20-line source build instead.

4. **Test fixture artifact** — will the `PendulumsFastSlowMimicTest` world fixture (ground collision at z=0) be adjusted upstream (raise the ground 0.5 mm or lock the base)? Otherwise a dartsim enablement PR would carry that 3/300-step noise in CI indefinitely.

5. **Parameterization** — the fix introduces ERP 0.4 and a correction cap of 10 rad/s as, apparently, hardcoded constants. Any intent to expose them as constraint parameters? Stiff, high-DoF coupled chains like our hand may need a different convergence/stiffness balance. (Also curious about the roadmap for the three "Not in this PR" items: Motor+Coupler mixed multi-DoF, self-mimic, and ball/free-joint chart coupling.)

6. **Recommended enablement path** — in September @scpeters noted that mimic joints were not enabled in the plugin due to the observed numerical instability and proposed a draft PR to evaluate it. With the DART-side root cause now fixed, what would you recommend for Harmonic-era users on the packaged stack (7.8.0 + 6.13.2): wait for a plugin wiring PR plus 6.20.1, or build DART `release-6.20` head with a local wiring in the meantime?

Whichever way this goes, we would be glad to test the wiring on the Harmonic line and report results here — we have a concrete 8-mimic-joint hand use case and the binary-level evidence trail from our Sep 21 comment.