gazebosim/gz-physics #442: Seg-faults in tests on macOS Monterey

scpeters, 2022-10-25:
## Environment
* OS Version: macOS Monterey 12.6.1-x86_64
* Source or binary build? source build of gz-physcis6 6.0.0

## Description
* Expected behavior: tests should pass, as they do on earlier versions of macOS
* Actual behavior: some tests seg-fault on macOS Monterey

## Steps to reproduce

1. On macOS Monterey:
2. `brew install gz-physics6 -s`
3. `brew test gz-physics6 --debug
4. When it fails, select option 5 to go to a shell
5. `lldb build/test`, then `r`, then `bt`

~~~
bash-3.2$ lldb ./test
(lldb) target create "./test"
Current executable set to '/private/tmp/gz-physics6-test-20221024-55214-ct6ppk/test' (x86_64).
(lldb) r
Process 55555 launched: '/private/tmp/gz-physics6-test-20221024-55214-ct6ppk/test' (x86_64)
Process 55555 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x1002e7528)
    frame #0: 0x00000001023a0ff7 libdart.6.12.2.dylib`std::__1::__hash_table<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::__unordered_map_hasher<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::__unordered_map_equal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::allocator<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::__deallocate_node(std::__1::__hash_node_base<std::__1::__hash_node<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, void*>*>*) + 47
libdart.6.12.2.dylib`std::__1::__hash_table<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::__unordered_map_hasher<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::__unordered_map_equal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::allocator<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::__deallocate_node:
->  0x1023a0ff7 <+47>: callq  *0x20(%rax)
    0x1023a0ffa <+50>: testb  $0x1, 0x10(%rbx)
    0x1023a0ffe <+54>: je     0x1023a1009               ; <+65>
    0x1023a1000 <+56>: movq   0x20(%rbx), %rdi
Target 0: (test) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x1002e7528)
  * frame #0: 0x00000001023a0ff7 libdart.6.12.2.dylib`std::__1::__hash_table<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::__unordered_map_hasher<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::__unordered_map_equal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::allocator<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::__deallocate_node(std::__1::__hash_node_base<std::__1::__hash_node<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, void*>*>*) + 47
    frame #1: 0x00000001023a0faa libdart.6.12.2.dylib`std::__1::__hash_table<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::__unordered_map_hasher<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::__unordered_map_equal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::allocator<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::~__hash_table() + 18
    frame #2: 0x00007ff80118bde4 libsystem_c.dylib`__cxa_finalize_ranges + 409
    frame #3: 0x00007ff80118bbfe libsystem_c.dylib`exit + 35
    frame #4: 0x00007ff80129f375 libdyld.dylib`dyld4::LibSystemHelpers::exit(int) const + 11
    frame #5: 0x0000000100029558 dyld`start + 504
(lldb)
~~~

## Output

~~~
The following tests FAILED:
	 11 - UNIT_FindFeatures_TEST (SEGFAULT)
	 36 - COMMON_TEST_joint_features_bullet-featherstone (SEGFAULT)
[... cut ...] _value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::~__hash_table() + 18
    frame #2: 0x00007ff80118bde4 libsystem_c.dylib`__cxa_finalize_ranges + 409
    frame #3: 0x00007ff80118bbfe libsystem_c.dylib`exit + 35
    frame #4: 0x00007ff80129f375 libdyld.dylib`dyld4::LibSystemHelpers::exit(int) const + 11
    frame #5: 0x00000001000c9558 dyld`start + 504
(lldb)
~~~

scpeters, 2022-10-25:
currently blocking monterey bottles for gz-physics6 in https://github.com/osrf/homebrew-simulation/pull/2089

j-rivero, 2022-10-25:
Could be something related to?

>  2022-10-24 20:49:39.444823-0700 INTEGRATION_DoublePendulum[56318:234810] dynamic_cast error 2: One or more of the following type_info's has hidden visibility or is defined in more than one translation unit. They should all have public visibility. N4urdf8GeometryE, N4urdf3BoxE, N4urdf6SphereE.

j-rivero, 2022-10-25:
re-reading seems more related to libdart-collision-ode.6.12.2 when using the CollisionDetector:

> frame #0: 0x00000001019b34c6 libdart-collision-ode.6.12.2.dylib`std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>::operator=(std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>&&) + 46

mjcarroll, 2022-10-31:
@scpeters are you looking into this?

scpeters, 2022-10-31:
> @scpeters are you looking into this?

I upgraded my laptop to Monterey, but I don't know how to proceed with this issue. I am not actively investigating it but am happy to test debugging ideas that people propose

scpeters, 2022-10-31:
some analysis of these build failures and backtraces:

## INTEGRATION_DoublePendulum

This test has only one test case, which finishes successfully and prints `[  PASSED  ] 1 test.` and then it seg-faults after that point.

## UNIT_FindFeatures_TEST

This test has multiple test cases, and the first one passes, but then it seg-faults during the second test case:

~~~
[ RUN      ] FindFeatures_TEST.ForwardStep
[       OK ] FindFeatures_TEST.ForwardStep (1948 ms)
[ RUN      ] FindFeatures_TEST.Unimplemented
Process 56311 stopped
~~~

## Analysis

This looks like an issue with shutdown? Or unloading the plugins?

Crola1702, 2023-05-15:
This is happening on mac.five.monterey (cloud machine). Reference build: [gz-physics6-homebrew/61 build](https://build.osrfoundation.org/job/ignition_physics-ci-gz-physics6-homebrew-amd64/61/)

These are the failing tests:
<details>

```
The following tests FAILED:
	 11 - UNIT_FindFeatures_TEST (SEGFAULT)
	 19 - COMMON_TEST_added_mass_dartsim (SEGFAULT)
	 27 - COMMON_TEST_addexternalforcetorque_dartsim (SEGFAULT)
	 35 - COMMON_TEST_basic_test_dartsim (SEGFAULT)
	 43 - COMMON_TEST_collisions_dartsim (SEGFAULT)
	 51 - COMMON_TEST_construct_empty_world_dartsim (SEGFAULT)
	 59 - COMMON_TEST_free_joint_features_dartsim (SEGFAULT)
	 65 - COMMON_TEST_joint_features_bullet-featherstone (SEGFAULT)
	 67 - COMMON_TEST_joint_features_dartsim (SEGFAULT)
	 75 - COMMON_TEST_joint_transmitted_wrench_features_dartsim (SEGFAULT)
	 83 - COMMON_TEST_kinematic_features_dartsim (SEGFAULT)
	 91 - COMMON_TEST_link_features_dartsim (SEGFAULT)
	 99 - COMMON_TEST_shape_features_dartsim (SEGFAULT)
	103 - COMMON_TEST_simulation_features_bullet (SEGFAULT)
	105 - COMMON_TEST_simulation_features_bullet-featherstone (SEGFAULT)
	107 - COMMON_TEST_simulation_features_dartsim (SEGFAULT)
	115 - COMMON_TEST_world_features_dartsim (SEGFAULT)
```

</details>

srmainwaring, 2023-05-18:
> This looks like an issue with shutdown? Or unloading the plugins?

Possibly a bit of both. Unloading the dart plugin causes a problem when exiting. Can reproduce with a minimal example:

```c++
int main(int argc, const char* argv[])
{
  plugin::Loader loader;
  PrimeTheLoader(loader);
}
```

where `PrimeTheLoader` is from `TestUtilities.hh`.

The `tpe` plugin does not cause an issue which can be verified by modifying that plugin's `CMakeLists.txt`to include:

```cmake
foreach(test UNIT_FindFeatures_TEST UNIT_RequestFeatures_TEST)
  if(TARGET ${test})
    target_compile_definitions(${test} PRIVATE
      "tpe_plugin_LIB=\"$<TARGET_FILE:${tpe_plugin}>\""
    )
  endif()
endforeach()
```

A work around might be to use the `nodelete=true` version of `gz::plugin::Loader::LoadLib()`, but the core problem seems to be with the dart library. It looks as though the destructor code for the `CollisionDetector` has been unloaded before the `Factory` goes out of scope (see details).

<details>

```bash
libdart.6.10.0.dylib!std::__1::__function::__value_func<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>::~__value_func() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__functional/function.h:464)
libdart.6.10.0.dylib!std::__1::__function::__value_func<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>::~__value_func() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__functional/function.h:462)
libdart.6.10.0.dylib!std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>::~function() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__functional/function.h:1169)
libdart.6.10.0.dylib!std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()>::~function() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__functional/function.h:1169)
libdart.6.10.0.dylib!std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >::~pair() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__utility/pair.h:42)
libdart.6.10.0.dylib!std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >::~pair() (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__utility/pair.h:42)
libdart.6.10.0.dylib!void std::__1::allocator_traits<std::__1::allocator<std::__1::__hash_node<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, void*> > >::destroy<std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, void, void>(std::__1::allocator<std::__1::__hash_node<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, void*> >&, std::__1::pair<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >*) (/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX13.1.sdk/usr/include/c++/v1/__memory/allocator_traits.h:319)
libdart.6.10.0.dylib!std::__1::__hash_table<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::__unordered_map_hasher<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::__unordered_map_equal<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> >, std::__1::equal_to<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, std::__1::hash<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > >, true>, std::__1::allocator<std::__1::__hash_value_type<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, std::__1::function<std::__1::shared_ptr<dart::collision::CollisionDetector> ()> > > >::__deallocate_node(std::__1::__hash_no [... cut ...]
libdart.6.10.0.dylib!dart::common::Factory<std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> >, dart::collision::CollisionDetector, std::__1::shared_ptr<dart::collision::CollisionDetector> >::~Factory() (/Volumes/MacPro2_DV1/Code/osrf/gz_garden_ws/src/dart/dart/common/Factory.hpp:80)
libsystem_c.dylib!__cxa_finalize_ranges (Unknown Source:0)
libsystem_c.dylib!exit (Unknown Source:0)
libdyld.dylib!dyld4::LibSystemHelpers::exit(int) const (Unknown Source:0)
dyld!start (Unknown Source:0)
```

</details>

Investigation ongoing...

scpeters, 2023-08-22:
while testing #529, I noticed a seg-fault in `COMMON_TEST_joint_features_bullet-featherstone` in the `JointFeaturesAttachDetachTest/0.JointAttachDetachSpawnedModel` test case. Here is a snippet of console output just before the failure and a backtrace from `lldb`:

~~~
[----------] 4 tests from JointFeaturesAttachDetachTest/0, where TypeParam = JointFeatureAttachDetachList
[ RUN      ] JointFeaturesAttachDetachTest/0.JointAttachDetach
JointFeaturesTest::GetLibToTest() lib/libgz-physics6-bullet-featherstone-plugin.dylib
Testing plugin: gz::physics::bullet_featherstone::Plugin
[       OK ] JointFeaturesAttachDetachTest/0.JointAttachDetach (133 ms)
[ RUN      ] JointFeaturesAttachDetachTest/0.JointAttachMultiple
JointFeaturesTest::GetLibToTest() lib/libgz-physics6-bullet-featherstone-plugin.dylib
Testing plugin: gz::physics::bullet_featherstone::Plugin
[       OK ] JointFeaturesAttachDetachTest/0.JointAttachMultiple (114 ms)
[ RUN      ] JointFeaturesAttachDetachTest/0.LinkCountsInJointAttachDetach
JointFeaturesTest::GetLibToTest() lib/libgz-physics6-bullet-featherstone-plugin.dylib
Testing plugin: gz::physics::bullet_featherstone::Plugin
[       OK ] JointFeaturesAttachDetachTest/0.LinkCountsInJointAttachDetach (113 ms)
[ RUN      ] JointFeaturesAttachDetachTest/0.JointAttachDetachSpawnedModel
JointFeaturesTest::GetLibToTest() lib/libgz-physics6-bullet-featherstone-plugin.dylib
Testing plugin: gz::physics::bullet_featherstone::Plugin
Process 93698 stopped
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x102656ba8)
    frame #0: 0x0000000101013f0f libBulletCollision.3.25.dylib`btCollisionWorld::~btCollisionWorld() + 41
libBulletCollision.3.25.dylib`btCollisionWorld::~btCollisionWorld:
->  0x101013f0f <+41>: movq   0xc8(%r12), %r15
    0x101013f17 <+49>: testq  %r15, %r15
    0x101013f1a <+52>: je     0x101013f57               ; <+113>
    0x101013f1c <+54>: movq   0x68(%r14), %rdi
Target 0: (COMMON_TEST_joint_features) stopped.
(lldb) bt
* thread #1, queue = 'com.apple.main-thread', stop reason = EXC_BAD_ACCESS (code=1, address=0x102656ba8)
  * frame #0: 0x0000000101013f0f libBulletCollision.3.25.dylib`btCollisionWorld::~btCollisionWorld() + 41
    frame #1: 0x0000000100dd861e libBulletDynamics.3.25.dylib`btMultiBodyDynamicsWorld::~btMultiBodyDynamicsWorld() + 14
    frame #2: 0x00000001060097dd libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::WorldInfo::~WorldInfo() [inlined] std::__1::default_delete<btMultiBodyDynamicsWorld>::operator(this=0x0000600003511c80, __ptr=<unavailable>)[abi:v15006](btMultiBodyDynamicsWorld*) const at unique_ptr.h:48:5 [opt]
    frame #3: 0x00000001060097d7 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::WorldInfo::~WorldInfo() [inlined] std::__1::unique_ptr<btMultiBodyDynamicsWorld, std::__1::default_delete<btMultiBodyDynamicsWorld> >::reset[abi:v15006](this=0x0000600003511c80, __p=0x0000000000000000) at unique_ptr.h:305:7 [opt]
    frame #4: 0x00000001060097c6 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::WorldInfo::~WorldInfo() [inlined] std::__1::unique_ptr<btMultiBodyDynamicsWorld, std::__1::default_delete<btMultiBodyDynamicsWorld> >::~unique_ptr[abi:v15006](this=0x0000600003511c80) at unique_ptr.h:259:19 [opt]
    frame #5: 0x00000001060097c6 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::WorldInfo::~WorldInfo() [inlined] std::__1::unique_ptr<btMultiBodyDynamicsWorld, std::__1::default_delete<btMultiBodyDynamicsWorld> >::~unique_ptr[abi:v15006](this=0x0000600003511c80) at unique_ptr.h:259:17 [opt]
    frame #6: 0x00000001060097c6 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::WorldInfo::~WorldInfo(this=0x0000600003511c48) at Base.hh:64:8 [opt]
    frame #7: 0x0000000106009b98 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::__shared_count::__release_shared[abi:v15006](this=0x0000600003511c30) at shared_ptr.h:174:9 [opt]
    frame #8: 0x0000000106009b7d libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::__shared_weak_count::__release_shared[abi:v15006](this=0x0000600003511c30) at shared_ptr.h:215:27 [opt]
    frame #9: 0x0000000106009b7d libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::shared_ptr<gz::physics::bullet_featherstone::WorldInfo>::~shared_ptr[abi:v15006](this=0x0000600000ca9d28) at shared_ptr.h:702:23 [opt]
    frame #10: 0x0000000106009b74 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::shared_ptr<gz::physics::bullet_featherstone::WorldInfo>::~shared_ptr[abi:v15006](this=0x0000600000ca9d28) at shared_ptr.h:700:5 [opt]
    frame #11: 0x0000000106009b74 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::pair<unsigned long const, std::__1::shared_ptr<gz::physics::bullet_featherstone::WorldInfo> >::~pair(this=0x0000600000ca9d20) at pair.h:40:29 [opt]
    frame #12: 0x0000000106009b74 libgz-physics6-bullet-featherstone-plugin.6.4.0.dylib`gz::physics::bullet_featherstone::Base::~Base() [inlined] std::__1::pair<unsigned long const, std::__1::shared_ptr<gz::physics: [... cut ...]  char const*) at gtest.cc:2635:14 [opt]
    frame #52: 0x000000010007733c COMMON_TEST_joint_features`testing::UnitTest::Run(this=0x00000001000ac080) at gtest.cc:5444:10 [opt]
    frame #53: 0x000000010000f0d8 COMMON_TEST_joint_features`main [inlined] RUN_ALL_TESTS() at gtest.h:2293:73 [opt]
    frame #54: 0x000000010000f0cb COMMON_TEST_joint_features`main(argc=<unavailable>, argv=0x00007ff7bfeff688) at joint_features.cc:1610:10 [opt]
    frame #55: 0x00007ff81242f41f dyld`start + 1903
(lldb)
~~~

azeey, 2023-08-22:
Repeating https://github.com/gazebosim/gz-physics/pull/529#issuecomment-1688350988 for visibility

BTW, I believe the segfaults in bullet-featherstone are due to the order of destruction of `unique_ptr`s in `Base.hh`. Bullet seems to have a lot of interdependencies, so deleting objects in the right order is tricky. I have been able to reproduce the problem on linux when running under Valgrind, but I haven't had the chance to fix the issue.

azeey, 2024-01-23:
@scpeters, I don't see any segfaults in https://build.osrfoundation.org/view/gz-garden/job/gz_physics-ci-gz-physics6-homebrew-amd64/ or https://build.osrfoundation.org/view/gz-harmonic/job/gz_physics-ci-gz-physics7-homebrew-amd64/. Can this issue be closed?

scpeters, 2024-01-23:
yeah, I think the tests have been fixed

PR #537 (2023-09-05): Fortress: backport fix for dartsim segfaults
# 🦟 Bug fix

Backport #529 to Fortress to fix #442 (gz-sim6 segfaults on macOS) on Fortress

## Summary

See #529 and #442 for context.

Original commit message:

Unregister collision detectors when the darstim plugin is unloaded (#529)

Fixes a segfault that occurs due to destructors being removed from memory before they're called. ---------

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has the [... cut]
closed 2024-01-23 by scpeters