gazebosim/gz-physics #449: Add Feature to allow runtime changes to joint friction, damping, spring properties

scpeters, 2022-11-15:
## Desired behavior

There are several joint parameters specified in `//joint/axis*/dynamics` such as `friction`, `damping`, `spring_stiffness`, and `spring_reference`.  Several of these parameters are supported by the `dartsim` plugin when [loading from an SDFormat file](https://github.com/gazebosim/gz-physics/blob/619625d1cd97aa4beaf046c5083fbda913530290/dartsim/src/SDFFeatures.cc#L131-L134), but there is no Feature for changing these parameters at runtime. Related to https://github.com/gazebosim/gz-physics/issues/96 which requested the ability to set joint limits at runtime, resolved in https://github.com/gazebosim/gz-physics/pull/260.

## Alternatives considered

## Implementation suggestion

See https://github.com/gazebosim/gz-physics/pull/260

## Additional context

francocipollone, 2024-04-10:
Hey there! One silly question. From what I've seen in the suggested implementation, adding this feature (and later implementation in gz-sim) shouldn't break api/abi, right? It should be able to easily backport it to fortress if needed, correct?

yaswanth1701, 2024-10-12:
Hi @scpeters, Can I work on this feature ?\

scpeters, 2024-10-26:
> Can I work on this feature ?

Yes, @yaswanth1701 you may work on this feature. 😁

Please target Ionic for your initial work and let me know if you have questions

yaswanth1701, 2024-10-31:
Sure.

yaswanth1701, 2024-11-19:
Hi @scpeters , I had a doubt is `setRestPosition` for spring same as `springreference` which you mentioned above. https://github.com/dartsim/dart/blob/98cf203390182bf5907c5f6912a258084e9f3758/dart/dynamics/Joint.hpp#L615-L650

yaswanth1701, 2024-11-22:
Hi @scpeters, I was thinking to write the test cases for the implementation of set of features to check if everything works well. Should I implement the `get` methods too ? And check if after few simulation iterations, values remains the same ? if possible can you please suggest possible tests.

scpeters, 2024-11-22:
> Hi @scpeters, I was thinking to write the test cases for the implementation of set of features to check if everything works well. Should I implement the `get` methods too ? And check if after few simulation iterations, values remains the same ? if possible can you please suggest possible tests.

yes, I think the get methods would be nice as well. The best test cases would use a physical expectation, for example, if you load a world with a pendulum initially at 90 degrees so that it would start oscillating as soon as the simulation starts; if you initialize the joint friction to a very high value, it should not move. then after a few steps, if you drop it to zero, it should start moving again. if you put it at a small value that allows some motion, it should decrease the total energy. that would be an ideal test, but it would be ok to just partially implement it

for joint damping, I would expect different levels of energy dissipation depending on the damping coefficient

for spring properties, you could start with a pendulum in a stable downward position then apply a spring stiffness with a reference position at 90 degrees and expect it to be pulled towards that reference position. if you add some damping, you can wait for it to come to an equilibrium and check the equilibrium point. you can also apply a constant effort or use a prismatic joint at an angle relative to the gravity vector in order to get a linear response instead of the nonlinear pendulum behavior

just some thoughts, please reach out if you want to discuss further

yaswanth1701, 2024-11-24:
> yes, I think the get methods would be nice as well. The best test cases would use a physical expectation, for example, if you load a world with a pendulum initially at 90 degrees so that it would [...]

Thanks a lot for the suggestions Steve. I will start implementing these tests. Also, what should be the expected behavior from the prismatic joint with constant force or gravity applied to it?

scpeters, 2024-11-24:
> Thanks a lot for the suggestions Steve. I will start implementing these tests. Also, what should be the expected behaviour from the prismatic joint with constant force or gravity applied to it?

if the parent link of the prismatic joint is the `world`, then I would use Newton's 2nd law to equate the sume of forces on the child link (gravity and the joint force) with the child link's linear acceleration. You only need to consider the forces and acceleration in the direction of the prismatic joint axis.

~~~
dot(sum_of_child_link_forces, joint_axis) = mass * dot(child_link_linear_acceleration, joint_axis)
sum_of_child_link_forces = mass * gravity_vector + joint_force_on_child_link
~~~

The the `gravity_vector` is constant, and the `joint_force_on_child_link` depends on the parameters applied; in general it is:

~~~
joint_force_on_child_link =
    - spring_stiffness * (joint_position - spring_reference)
    - damping * joint_velocity
    - joint_friction * sign(joint_velocity)
~~~

With no stiffness or damping, I would expect two regimes of behavior for a child of a prismatic joint subject to joint friction:

* when the joint friction is larger than `mass * abs(dot(gravity_vector, joint_axis))`, the child link should not move (joint_velocity = 0)
* when the joint friction is smaller than `mass * abs(dot(gravity_vector, joint_axis))`, the child link will have constant linear acceleration along the joint axis, with two possible values depending on the joint_velocity direction compared to the gravity direction: `child_link_linear_acceleration = dot(gravity_vector, joint_axis) joint_friction / mass * sign(joint_velocity)`

I would use a similar approach to predict the behavior with spring and damping properties.

yaswanth1701, 2024-11-26:
Thanks a lot for the detailed explanation Steve 😁. I will start with implementation.

PR #723 (2025-04-30): Features to change joint properties at run-time

# 🎉 New feature

Closes #449 

## Summary
Added features to change joint spring-stiffness, damping, friction, and rest position at run-time.  

## Test it
To test the added feature please run

``` bash
mkdir build
cd build
make 
make test
```

## Checklist
- [X] Signed all commits for DCO
- [X] Added tests
- [ ] Added example and/or tutorial
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebo [... cut]
closed 2025-04-30 by scpeters