gazebosim/gz-physics #450: Errors about Heightmap and Mesh construction

azeey, 2022-11-17:
## Environment
* OS Version: Ubuntu 20.04 
* Source or binary build? source, gz-physics6

## Description
* Expected behavior: No errors printed
* Actual behavior:  The following errors are printed
```
[Err] [SDFFeatures.cc:309] Heightmap construction from an SDF has not been implemented yet for dartsim.
[Err] [SDFFeatures.cc:320] Mesh construction from an SDF has not been implemented yet for dartsim.
```

## Steps to reproduce

1. Run `gz sim -v3`  with any world that contains a heightmap or a mesh.

DatSpace, 2022-11-18:
I can verify I have the same issue after upgrading my packages.

`[Err] [SDFFeatures.cc:320] Mesh construction from an SDF has not been implemented yet for dartsim.
`

With the same sdf, this issue was not there before updating with apt upgrade. The last apt upgrade was probably about a month ago.

I am currently running on Ubuntu 22.04, Gazebo garden, on the latest apt release.

azeey, 2022-11-18:
The errors appear to be harmless, but I think they started after https://github.com/gazebosim/gz-sim/pull/1560