gazebosim/gz-physics #463: `double free or corruption (out)` when running with `gazebo-forks/dart`

azazdeaz, 2022-12-18:
Hi There,

I'm trying to run Gazebo (Garden) with the [gazebo-forks version of dart](https://github.com/gazebo-forks/dart)  
When i run `gz sim` with gz-physics built with `gazebo-forks/dart` it prints `double free or corruption (out)` and the physics are not working.  
You can run this Dockerfile to reproduce the issue:
[Dockerfile.humble.dart-fork.txt](https://github.com/gazebosim/gz-physics/files/10254569/Dockerfile.humble.dart-fork.txt)

Interestingly, i have a distrobox also based on `osrf/ros:humble-desktop` which can build working gz-physics binaries but i can't reproduce this with a fresh Dockerfile :sweat_smile:  I can't find what is the difference: `apt list --installed` is technically the same, cmake seem to have the same result for both `dart` and `gz-physics` 

What do you think i could check/try to find the issue?

azeey, 2023-03-02:
I wonder if this has been fixed by https://github.com/gazebosim/gz-physics/pull/480. Do you mind trying it again?

azeey, 2023-03-13:
Testing with Dockerfile you provided, looks like #480 doesn't fix the issue.