gazebosim/gz-physics #483: :farmer: `SimulationFeaturesFallingTest` and `JointFeaturesAttachDetachTest` failing when run on Windows Nuc

Crola1702, 2023-03-08:
## Environment
* OS Version: Windows
* Source Gazebo Garden

## Description
* Expected behavior: All test should pass
* Actual behavior: 4 tests from `SimulationFeaturesFallingTest` test suite fail when built on Windows nuc

## Steps to reproduce
1. Run a build in [gz-physics6 job](https://build.osrfoundation.org/job/ign_physics-gz-6-win/) on Windows Nuc
2. See tests fail.

## Output

Reference build: https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/

Test regressions:
* [JointFeaturesAttachDetachTest/0.JointAttachMultiple](https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/testReport/(root)/JointFeaturesAttachDetachTest_0/JointAttachMultiple_2/)
* [SimulationFeaturesStepTest/0.StepWorld](https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/testReport/(root)/SimulationFeaturesStepTest_0/StepWorld_2/)
* [SimulationFeaturesFallingTest/0.Falling](https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/testReport/(root)/SimulationFeaturesFallingTest_0/Falling_2/)
* [SimulationFeaturesShapeFeaturesTest/0.ShapeFeatures](https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/testReport/(root)/SimulationFeaturesShapeFeaturesTest_0/ShapeFeatures_2/)

Log output:
```
C:\Jenkins\workspace\ign_physics-gz-6-win\ws\gz-physics\test\common_test\joint_features.cc:1288
The difference between 0.0 and body2LinearVelocity.Z() is 0.010009272024035454, which exceeds 1e-3, where
0.0 evaluates to 0,
body2LinearVelocity.Z() evaluates to -0.010009272024035454, and
1e-3 evaluates to 0.001.
...
unknown file
C++ exception with description "Access violation - no RTTI data!" thrown in the test body.
...
unknown file
SEH exception with code 0xc0000005 thrown in the test body.
...
unknown file
C++ exception with description "Access violation - no RTTI data!" thrown in the test body.
```

* These tests are failing only on Windows Nuc
  * [Beefy green build](https://build.osrfoundation.org/job/ign_physics-gz-6-win/53/)
  * [Nuc yellow build](https://build.osrfoundation.org/job/ign_physics-gz-6-win/52/)

From the `COMMON_TEST_simulation_features_bullet` test suite: Most of the tests in Beefy are skipped, however, the only 2 that run, fail.
* SimulationFeaturesFallingTest/0.Falling: is skipped on Beefy, but not on Nuc
  * The failure in Nuc shows: `unknown file: error: C++ exception with description "Access violation - no RTTI data!" thrown in the test body.` (Related to https://github.com/osrf/buildfarmer/issues/321)
* SimulationFeaturesStepTest/0.StepWorld: passes in Beefy, but not on Nuc
  * The failure in Nuc shows: `unknown file: error: SEH exception with code 0xc0000005 thrown in the test body.` (Related to https://github.com/osrf/buildfarmer/issues/320)

In conclusion: Weird Windows problems here

scpeters, 2023-07-18:
I've documented some investigation of the SEH exception issues in https://github.com/gazebosim/gz-physics/issues/520

Crola1702, 2024-04-08:
These regressions haven't happened since 2024-08-02. I'm closing this for now

closed 2024-04-08 by Crola1702