gazebosim/gz-physics #558: Build fail if sdformat is missing even if it is marked as optional dependency

traversaro, 2023-09-30:
## Environment
* OS Version: 22.04
* Source or binary build? Source build of 7.0.0

## Description
* Expected behavior: If the CMake configuration passes, I imagine the build should pass. 
* Actual behavior: CMake config passes claiming that some components are disabled as sdformat14 is missing, but then the build fails.
* 
## Steps to reproduce

1. Build gz-physics 7 without sdformat14
2.
3.

## Output

~~~
2023-09-29T23:58:53.9694299Z -- The C compiler identification is GNU 12.3.0
2023-09-29T23:58:54.0992704Z -- The CXX compiler identification is GNU 12.3.0
2023-09-29T23:58:54.1391192Z -- Detecting C compiler ABI info
2023-09-29T23:58:54.2402086Z -- Detecting C compiler ABI info - done
2023-09-29T23:58:54.2668992Z -- Check for working C compiler: /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_build_env/bin/x86_64-conda-linux-gnu-cc - skipped
2023-09-29T23:58:54.2679798Z -- Detecting C compile features
2023-09-29T23:58:54.2696071Z -- Detecting C compile features - done
2023-09-29T23:58:54.2888262Z -- Detecting CXX compiler ABI info
2023-09-29T23:58:54.3921221Z -- Detecting CXX compiler ABI info - done
2023-09-29T23:58:54.4196412Z -- Check for working CXX compiler: /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_build_env/bin/x86_64-conda-linux-gnu-c++ - skipped
2023-09-29T23:58:54.4203725Z -- Detecting CXX compile features
2023-09-29T23:58:54.4216522Z -- Detecting CXX compile features - done
2023-09-29T23:58:54.4557404Z -- gz-physics7 version 7.0.0
2023-09-29T23:58:54.4558444Z -- Operating system is Linux
2023-09-29T23:58:54.4656447Z -- 
2023-09-29T23:58:54.4656902Z 
2023-09-29T23:58:54.4657184Z -- ====== Finding Dependencies ======
2023-09-29T23:58:54.4694746Z -- Looking for gz-common5 -- found version 5.4.2
2023-09-29T23:58:54.4710428Z -- Searching for dependencies of gz-common5
2023-09-29T23:58:54.4739273Z -- Looking for gz-utils2 -- found version 2.1.0
2023-09-29T23:58:54.4744490Z -- Searching for dependencies of gz-utils2
2023-09-29T23:58:54.4788166Z -- Looking for dlfcn.h - found
2023-09-29T23:58:54.4854245Z -- Looking for libdl - found
2023-09-29T23:58:54.4893192Z -- Found DL: TRUE  
2023-09-29T23:58:54.5184088Z -- Checking for module 'uuid'
2023-09-29T23:58:54.6343455Z --   Found uuid, version 2.38.1
2023-09-29T23:58:54.9484838Z -- Found UUID: TRUE  
2023-09-29T23:58:54.9550020Z -- Searching for <gz-common5> component [geospatial]
2023-09-29T23:58:54.9631204Z -- Looking for gz-common5-geospatial -- found version 5.4.2
2023-09-29T23:58:54.9639676Z -- Searching for dependencies of gz-common5-geospatial
2023-09-29T23:58:54.9681007Z -- Looking for gz-math7 -- found version 7.3.0
2023-09-29T23:58:54.9687979Z -- Searching for dependencies of gz-math7
2023-09-29T23:58:54.9699953Z -- Looking for gz-utils2 -- found version 2.1.0
2023-09-29T23:58:54.9735761Z -- Looking for gz-common5 -- found version 5.4.2
2023-09-29T23:58:54.9742337Z -- Searching for dependencies of gz-common5
2023-09-29T23:58:54.9753296Z -- Looking for gz-utils2 -- found version 2.1.0
2023-09-29T23:58:54.9760656Z -- Looking for dlfcn.h - found
2023-09-29T23:58:54.9765035Z -- Looking for libdl - found
2023-09-29T23:58:55.0088595Z -- Searching for <gz-common5> component [graphics]
2023-09-29T23:58:55.0122442Z -- Looking for gz-common5-graphics -- found version 5.4.2
2023-09-29T23:58:55.0130020Z -- Searching for dependencies of gz-common5-graphics
2023-09-29T23:58:55.0141795Z -- Looking for gz-math7 -- found version 7.3.0
2023-09-29T23:58:55.0226917Z -- Found GzAssimp: TRUE (found version "5.2.4") 
2023-09-29T23:58:55.0265288Z -- Searching for <gz-common5> component [graphics]
2023-09-29T23:58:55.0265795Z -- Searching for <gz-common5> component [profiler]
2023-09-29T23:58:55.0297917Z -- Looking for gz-common5-profiler -- found version 5.4.2
2023-09-29T23:58:55.0302842Z -- Searching for dependencies of gz-common5-profiler
2023-09-29T23:58:55.0320753Z -- Looking for gz-common5 - found
2023-09-29T23:58:55.0321005Z 
2023-09-29T23:58:55.0346819Z -- Looking for gz-math7 -- found version 7.3.0
2023-09-29T23:58:55.0353284Z -- Searching for dependencies of gz-math7
2023-09-29T23:58:55.0363685Z -- Looking for gz-utils2 -- found version 2.1.0
2023-09-29T23:58:55.0369186Z -- Searching for <gz-math7> component [eigen3]
[... 146 lines cut ...]
2023-09-29T23:59:01.3684092Z [3/107] Building CXX object src/CMakeFiles/gz-physics7.dir/CompositeData.cc.o
2023-09-29T23:59:01.6480685Z [4/107] Building CXX object src/CMakeFiles/gz-physics7.dir/FrameID.cc.o
2023-09-29T23:59:01.6664443Z [5/107] Building CXX object src/CMakeFiles/gz-physics7.dir/DataStatusMask.cc.o
2023-09-29T23:59:02.5678867Z [6/107] Building CXX object src/CMakeFiles/gz-physics7.dir/Identity.cc.o
2023-09-29T23:59:02.8627051Z [7/107] Building CXX object src/CMakeFiles/gz-physics7.dir/gz-physics7_get_install_prefix_impl.cc.o
2023-09-29T23:59:07.6277807Z [8/107] Building CXX object src/CMakeFiles/UNIT_Cloneable_TEST.dir/Cloneable_TEST.cc.o
2023-09-29T23:59:10.1845347Z [9/107] Building CXX object src/CMakeFiles/gz-physics7.dir/InstallationDirectories.cc.o
2023-09-29T23:59:10.4149109Z [10/107] Linking CXX shared library lib/libgz-physics7.so.7.0.0
2023-09-29T23:59:10.4466224Z [11/107] Creating library symlink lib/libgz-physics7.so.7 lib/libgz-physics7.so
2023-09-29T23:59:14.7194223Z [12/107] Building CXX object src/CMakeFiles/UNIT_Feature_TEST.dir/Feature_TEST.cc.o
2023-09-29T23:59:14.9628178Z [13/107] Building CXX object src/CMakeFiles/UNIT_CompositeData_TEST.dir/CompositeData_TEST.cc.o
2023-09-29T23:59:15.4366168Z [14/107] Building CXX object src/CMakeFiles/UNIT_FeatureList_TEST.dir/FeatureList_TEST.cc.o
2023-09-29T23:59:17.4350756Z [15/107] Building CXX object src/CMakeFiles/UNIT_FilterTuple_TEST.dir/FilterTuple_TEST.cc.o
2023-09-29T23:59:19.5657659Z [16/107] Building CXX object src/CMakeFiles/UNIT_FindFeatures_TEST.dir/FindFeatures_TEST.cc.o
2023-09-29T23:59:21.7093520Z [17/107] Building CXX object test/gtest_vendor/CMakeFiles/gtest_main.dir/src/gtest_main.cc.o
2023-09-29T23:59:22.4554940Z [18/107] Building CXX object test/common_test/CMakeFiles/COMMON_TEST_added_mass.dir/added_mass.cc.o
2023-09-29T23:59:22.4557395Z FAILED: test/common_test/CMakeFiles/COMMON_TEST_added_mass.dir/added_mass.cc.o 
2023-09-29T23:59:22.4573638Z /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_build_env/bin/x86_64-conda-linux-gnu-c++ -DGZ_PHYSICS_RESOURCE_DIR=\"/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/resources\" -DGZ_PROFILER_ENABLE=0 -DTEST_WORLD_DIR=\"/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/test/common_test/worlds/\" -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/include -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/build/include -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold/include/bullet -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/build/core/include -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/mesh/include -I/home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/build/mesh/include -isystem /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold/include/gz/plugin2 -isystem /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold/include/gz/utils2 -isystem /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold/include/gz/common5 -isystem /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/_h_env_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placehold_placeho [... cut ...]
2023-09-29T23:59:22.4580443Z /home/conda/feedstock_root/build_artifacts/gz-physics7_1696031713776/work/test/common_test/added_mass.cc:33:10: fatal error: gz/physics/sdf/ConstructWorld.hh: No such file or directory
2023-09-29T23:59:22.4580986Z    33 | #include <gz/physics/sdf/ConstructWorld.hh>
2023-09-29T23:59:22.4581315Z       |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
2023-09-29T23:59:22.4581628Z compilation terminated.
2023-09-29T23:59:26.9355084Z [19/107] Building CXX object src/CMakeFiles/UNIT_SpecifyData_TEST.dir/SpecifyData_TEST.cc.o
2023-09-29T23:59:32.5215093Z [20/107] Building CXX object test/gtest_vendor/CMakeFiles/gtest.dir/src/gtest-all.cc.o
~~~

traversaro, 2023-09-30:
Perhaps sdformat14 should be a compulsory dep at this point?

scpeters, 2023-10-30:
I attempted to fix some similar issues in https://github.com/gazebosim/gz-physics/pull/523, but I guess there's still a bit more to be done

scpeters, 2023-10-30:
see also https://github.com/gazebosim/gz-physics/issues/522