gazebosim/gz-physics #565: bullet-featherstone: F/T sensor reports incorrect values at fixed joint if joint forces are applied

shameekganguly, 2023-11-01:
Repro steps:
1. Create a model with a few revolute joints and links.
2. At the last link, add another (payload) link with a fixed joint.
3. Place an F/T sensor at the fixed joint.
4. Add a joint position controller system to control the position of one of the revolute joints using joint force commands (PID).
5. Start the simulation and wait for the model to reach steady state.
6. Check published F/T sensor value. 

Expected value of force: weight of the attached payload link in the sensor frame.
Actual value of force: different from above, and magnitude is larger than the weight of the payload link.

You can test this with this sdf file: 
[fixed_joint_ft_sensor_bullet.sdf.txt](https://github.com/gazebosim/gz-physics/files/13223378/fixed_joint_ft_sensor_bullet.sdf.txt). Run gz sim and wait for steady state. Then check `/ft_sensor/wrench` topic for F/T values. Expected force is `[0 0 -9.8]`, but the published value is `[0 -0.34 -13.85]`.

Note that the bug only occurs when using bullet-featherstone and not for dartsim. Also, if a joint velocity command is used instead of joint force command to control the joint position (uncomment the second joint position controller plugin in the sdf), then the bug does not occur.

azeey, 2023-11-01:
Is this related to https://github.com/bulletphysics/bullet3/issues/4451? Perhaps we can use the python file there to test without gz-physics to see if it's an issue in Bullet or gz-physics.

shameekganguly, 2023-11-01:
> Is this related to [bulletphysics/bullet3#4451](https://github.com/bulletphysics/bullet3/issues/4451)? Perhaps we can use the python file there to test without gz-physics to see if it's an issue in Bullet or gz-physics.

Its possible that this is a related issue in bullet. I'll try to test with the python file this week before I am OOO.

iche033, 2023-11-04:
looks like the bug could be related to https://github.com/bulletphysics/bullet3/issues/2966. I tested bullet (`master` branch) with the patch posted in https://github.com/bulletphysics/bullet3/issues/2966#issuecomment-688136359 and I

scpeters, 2023-11-06:
I added this to the tracking issue in #545

iche033, 2024-04-17:
fixed upstream with https://github.com/bulletphysics/bullet3/pull/4583

closed 2024-04-17 by iche033