gazebosim/gz-physics #612: `DetachableJointTest.CorrectAttachmentPoints` dartsim test fails on Homebrew CI

iche033, 2024-03-29:
The `DetachableJointTest.CorrectAttachmentPoints` dartsim test started failing. This is likely caused by the new ODE 0.16.5 bottle, see https://github.com/gazebosim/gz-physics/pull/611#issuecomment-2027523960

## Environment
* OS Version: macOS homebrew
* Source or binary build? `gz-physics7` and `main` branches on homebrew jenkins 

## Description
* Expected behavior: All tests pass on homebrew
* Actual behavior:  `DetachableJointTest.CorrectAttachmentPoints` fails a position check after detaching joint (dartsim plugin, homebrew)

## Output

Test failure output from homebrew jenkins:

```
59: Testing plugin: gz::physics::dartsim::Plugin
59: /Users/jenkins/jenkins-agent/workspace/gz_physics-ci-pr_any-homebrew-amd64/gz-physics/test/common_test/detachable_joint.cc:190: Failure
59: The difference between 0.15 and frameDataC1L1.pose.translation().z() is 0.10148802840906343, which exceeds 1e-2, where
59: 0.15 evaluates to 0.14999999999999999,
59: frameDataC1L1.pose.translation().z() evaluates to 0.048511971590936565, and
59: 1e-2 evaluates to 0.01.
```

scpeters, 2024-03-29:
potential fix in https://github.com/Homebrew/homebrew-core/pull/167521

scpeters, 2024-04-02:
Actual fix in https://github.com/Homebrew/homebrew-core/pull/167721, reverting the workaround in https://github.com/gazebosim/gz-physics/pull/615

closed 2024-04-11 by scpeters