gazebosim/gz-physics #614: Update documentation for physics integrations with bullet

aagrawal05, 2024-03-30:
Hi, I'm a new contributor and wanted to get into understanding the various systems through contributing.

## Desired behavior

Update `03_physics_plugins.md` to include comparison with the bullet engine—resolving the TODO on line 76.

## Alternatives considered

## Implementation suggestion

## Additional context

I thought this might be a good first contributor issue. Please let me know if you have any other documentation or entry level tasks that need doing.

aagrawal05, 2024-03-30:
Accidentally closed...

azeey, 2024-04-03:
@aagrawal05 Thanks for volunteering to contribute. Are you specifically interested in gz-physics or would you consider other Gazebo libraries as well? I ask because the gz-physics codebase might not be great for a first time contributor unless you have a strong background in C++.

aagrawal05, 2024-04-03:
Hi @azeey, my pleasure.

Currently, I have an intermediate level of understanding in C++. I'm interested in contributing to Gazebo in general—I'm already working on another first issue in gazebosim/sdformat.

That said, once I am more familiar, I would definitely be interested in contributing more in-depth to the gz-physics library as I have some features I would like to implement relating to underwater robotics (inspired by the [uuv_simulator](https://github.com/uuvsimulator/uuv_simulator)). But that's a long-term goal.

In any case, I would like to try and navigate the codebase and get a feel for the architecture through this documentation task.