gazebosim/gz-physics #626: :farmer: `DetachableJointTest.CorrectAttachmentPoints` test failing consistently in gz-physics6-homebrew

Crola1702, 2024-04-18:
## Environment
* OS Version: MacOS (Ventura and Monterey)
* Source build (Garden and Main)

## Description
* Expected behavior: Tests should pass
* Actual behavior: [DetachableJointTest/0.CorrectAttachmentPoints](https://build.osrfoundation.org/job/gz_physics-ci-gz-physics6-homebrew-amd64/35/testReport/junit/(root)/DetachableJointTest_0/CorrectAttachmentPoints/) is failing in gz_physics6

## Steps to reproduce
1. Run a build
2. See test fail

## Output

Reference builds: 
* https://build.osrfoundation.org/job/gz_physics-ci-gz-physics6-homebrew-amd64/35/

Failing test regression: [DetachableJointTest/0.CorrectAttachmentPoints](https://build.osrfoundation.org/job/gz_physics-ci-gz-physics6-homebrew-amd64/35/testReport/junit/(root)/DetachableJointTest_0/CorrectAttachmentPoints/)
```
/Users/jenkins/jenkins-agent/workspace/gz_physics-ci-gz-physics6-homebrew-amd64/gz-physics/test/common_test/detachable_joint.cc:190
The difference between 0.15 and frameDataC1L1.pose.translation().z() is 0.10148802840906343, which exceeds 1e-2, where
0.15 evaluates to 0.14999999999999999,
frameDataC1L1.pose.translation().z() evaluates to 0.048511971590936565, and
1e-2 evaluates to 0.01.

```

Test hasn't been modified for 6 months. It was added in: https://github.com/gazebosim/gz-physics/pull/530

Could be related to https://github.com/gazebosim/gz-physics/issues/620?

scpeters, 2024-04-18:
this test has been disabled in https://github.com/gazebosim/gz-physics/pull/621, which should be backported to `gz-physics6`

azeey, 2024-05-06:
@Crola1702 would you be able to do the backport or do you need @scpeters's help?

scpeters, 2024-05-07:
> this test has been disabled in #621, which should be backported to `gz-physics6`

sorry I got mixed up. This issue is a duplicate of https://github.com/gazebosim/gz-physics/issues/612, which was fixed by https://github.com/Homebrew/homebrew-core/pull/167721

no more backports are needed here

closed 2024-05-07 by scpeters