gazebosim/gz-physics #627: :farmer: :farmer: `JointFeaturesDetachTest/0.JointDetach` test failing consistently in gz-physics-main-homebrew

Crola1702, 2024-04-18:
## Environment
* OS Version: MacOS (Ventura and Monterey)
* Source build: Main

## Description
* Expected behavior: Tests should pass
* Actual behavior: [JointFeaturesDetachTest/0.JointDetach](https://build.osrfoundation.org/job/gz_physics-ci-main-homebrew-amd64/38/testReport/junit/(root)/JointFeaturesDetachTest_0/JointDetach_2/)

## Steps to reproduce
1. Run a build
2. See test fail

## Output

Reference builds: 
* https://build.osrfoundation.org/job/gz_physics-ci-main-homebrew-amd64/38/

Failing test regression: [JointFeaturesDetachTest/0.JointDetach](https://build.osrfoundation.org/job/gz_physics-ci-main-homebrew-amd64/38/testReport/junit/(root)/JointFeaturesDetachTest_0/JointDetach_2/)

```
/Users/jenkins/jenkins-agent/workspace/gz_physics-ci-main-homebrew-amd64/gz-physics/test/common_test/joint_features.cc:969
The difference between 0.0 and upperLinkLinearVelocity.X() is 0.00035244933516466117, which exceeds 1e-6, where
0.0 evaluates to 0,
upperLinkLinearVelocity.X() evaluates to -0.00035244933516466117, and
1e-6 evaluates to 9.9999999999999995e-07.
```

Might be related to https://github.com/gazebosim/gz-physics/issues/620

scpeters, 2024-04-18:
this test was disabled in https://github.com/gazebosim/gz-physics/pull/621, which is being merged forward in https://github.com/gazebosim/gz-physics/pull/624

Crola1702, 2024-04-22:
As the last comment said. This was fixed

closed 2024-04-22 by Crola1702