gazebosim/gz-physics #634: dartsim: Support joint dynamics parameters on Ball joints

azeey, 2024-05-08:
## Desired behavior
Support for parameters such as damping on ball joints.

## Alternatives considered
none

## Implementation suggestion
Currently, none of the dynamics parameters are supported. The comment on https://github.com/gazebosim/gz-physics/blob/62ee52cf00c39b50eeea3570f79623afa2225fc8/dartsim/src/SDFFeatures.cc#L1120 indicates that this is because SDFormat doesn't support it and the following issues confirm that:

https://github.com/gazebosim/sdformat/issues/331
https://github.com/gazebosim/sdformat/issues/1250
https://github.com/gazebosim/sdformat/issues/860

However, maybe we can add an implementation where the parameters from the first `<axis>` are used for all the DOFs.

## Additional context
none

vbalaji21, 2024-05-13:
Can you please let me know when will this feature be implemented ? Because I am looking for a 3 DOF rotational joint with damping. I am not able to have any work around for this problem as I am unable to have 3 rotational joints on the same spot.

azeey, 2024-07-15:
I've added the `help wanted` label. We are currently not planning to work on this in the near term, but would accept pull requests.

PR #753 (2025-07-18): BallJoint: apply axis properties to ball joints
# 🎉 New feature

Closes #634

## Summary

Allow joint properties to be applied to ball joints in the dart sim physics engine.

As suggested in https://github.com/gazebosim/gz-physics/issues/634, the properties from the first `<axis>` field in the ball joint are applied to all DoF if this element is set. Otherwise the defaults are used, which is current behaviour.

## Test it

- Add an `<axis>` element to a model using a ball joint and set some of the properties (eg. `<limit>`).
- Prior to the change the property would have no effect, after the ch [... cut]