gazebosim/gz-physics #635: Velocity decay for links not working

vbalaji21, 2024-05-08:
## Environment
* OS Version:  Ubuntu 20.04 
* Source or binary build? Ignition Fortress

## Description
* Expected behavior: Damping action of links based on velocity decay tags
* Actual behavior: Damping does not happen

## Steps to reproduce

1. Make a simple 2 link manipulator (base link and another free link at an intial pose to enable free falling under gravity.
2. Add velocity decay tag for each link (both linear and angular with a nice coefficent eg:- 0.04)
3. On pressing play simulation the free link should fall down under the effect of gravity and come to rest position equilibrium, It does not happen as it is always moving nothing to damp this motion. 

## Output
I have attached a video and SDF file for this.

https://github.com/gazebosim/gz-physics/assets/14769004/7f4defb3-00b0-4ad5-aa9c-db654f164f1b

`<?xml version="1.0" encoding="UTF-8"?>

<sdf version="1.6">
   <world name="multicopter">
      <physics name="4ms" type="ignored">
         <max_step_size>0.004</max_step_size>
         <real_time_factor>1.0</real_time_factor>
      </physics>
      <plugin filename="ignition-gazebo-physics-system" name="gz::sim::systems::Physics" />
      <plugin filename="ignition-gazebo-scene-broadcaster-system" name="gz::sim::systems::SceneBroadcaster" />
      <plugin filename="ignition-gazebo-user-commands-system" name="gz::sim::systems::UserCommands" />
      <plugin filename="ignition-gazebo-sensors-system" name="gz::sim::systems::Sensors">
         <render_engine>ogre2</render_engine>
      </plugin>
      <light type="directional" name="sun">
         <cast_shadows>true</cast_shadows>
         <pose>0 0 10 0 0 0</pose>
         <diffuse>0.8 0.8 0.8 1</diffuse>
         <specular>0.2 0.2 0.2 1</specular>
         <attenuation>
            <range>1000</range>
            <constant>0.9</constant>
            <linear>0.01</linear>
            <quadratic>0.001</quadratic>
         </attenuation>
         <direction>-0.5 0.1 -0.9</direction>
      </light>
      <model name="ground_plane">
         <static>true</static>
         <link name="link">
            <collision name="collision">
               <geometry>
                  <plane>
                     <normal>0 0 1</normal>
                     <size>100 100</size>
                  </plane>
               </geometry>
            </collision>
            <visual name="visual">
               <geometry>
                  <plane>
                     <normal>0 0 1</normal>
                     <size>100 100</size>
                  </plane>
               </geometry>
               <material>
                  <ambient>0.8 0.8 0.8 1</ambient>
                  <diffuse>0.8 0.8 0.8 1</diffuse>
                  <specular>0.8 0.8 0.8 1</specular>
               </material>
            </visual>
         </link>
      </model>
      <model name="connected_system">
         <include>
            <name>pole_1</name>
            <pose>0 0 1.5 0 0 0</pose>
            <uri>/home/vignesh/.ignition/fuel/custom_models/pole</uri>
         </include>

         <link name="link1">
            <pose>0.0 -0.1 3.0 1.57 0 0</pose>
            <velocity_decay>
               <linear>2.1</linear>
               <angular>2.1</angular>
            </velocity_decay>
            <inertial>
               <inertia>
                  <ixx>0.3358e-03</ixx>
                  <ixy>0.0</ixy>
                  <ixz>0.0</ixz>
                  <iyy>0.3358e-03</iyy>
                  <iyz>0.0</iyz>
                  <izz>5.0e-06</izz>
               </inertia>
               <mass>0.001</mass>
            </inertial>
            <collision name="collision">
               <geometry>
                  <cylinder>
                     <length>0.2</length>
                     <radius>0.01</radius>
                  </cylinder>
               </geometry>
            </collision>
            <visual name="visual">
               <geometry>
                  <cylinder>
                     <length>0.2</length>
                     <radius>0.01</radius>
                  </cylinder>
               </geometry>
               <material>
                  <script>
                     <uri>file://media/materials/scripts/gazebo.material</uri>
                     <name>Gazebo/Black</name>
                  </script>
               </material>
            </visual>
         </link>
         
         <joint type="ball" name="hinge_1">
            <pose>0 0 1.5 0 0 0</pose>
            <parent>link1</parent>
            <child>pole_1::p1</child>
            <axis>
               <use_parent_model_frame>true</use_parent_model_frame>
               <xyz>1 1 1</xyz>
               <dynamics>
                  <damping>10.04</damping>
               </dynamics>
            </axis>
         </joint>         

      </model>
   </world>
</sdf>
`

azeey, 2024-05-08:
Thanks for the SDF file, but as is, we won't be able to try it because it references a file (            `<uri>/home/vignesh/.ignition/fuel/custom_models/pole</uri>`) that we won't have. Do you mind simplifying it, so it's all in just one model. Alternatively, paste the contents of the referenced file where the `<include>` is?

vbalaji21, 2024-05-08:
@azeey Yes, I have simplified and pasted the entire contents of the sdf in a simple way (I have also tested it again to reproduce it).

`<?xml version="1.0" encoding="UTF-8"?>
<sdf version="1.6">
   <world name="multicopter">
      <physics name="4ms" type="ignored">
         <max_step_size>0.004</max_step_size>
         <real_time_factor>1.0</real_time_factor>
      </physics>
      <plugin filename="ignition-gazebo-physics-system" name="gz::sim::systems::Physics" />
      <plugin filename="ignition-gazebo-scene-broadcaster-system" name="gz::sim::systems::SceneBroadcaster" />
      <plugin filename="ignition-gazebo-user-commands-system" name="gz::sim::systems::UserCommands" />
      <plugin filename="ignition-gazebo-sensors-system" name="gz::sim::systems::Sensors">
         <render_engine>ogre2</render_engine>
      </plugin>
      <light type="directional" name="sun">
         <cast_shadows>true</cast_shadows>
         <pose>0 0 10 0 0 0</pose>
         <diffuse>0.8 0.8 0.8 1</diffuse>
         <specular>0.2 0.2 0.2 1</specular>
         <attenuation>
            <range>1000</range>
            <constant>0.9</constant>
            <linear>0.01</linear>
            <quadratic>0.001</quadratic>
         </attenuation>
         <direction>-0.5 0.1 -0.9</direction>
      </light>
      <model name="ground_plane">
         <static>true</static>
         <link name="link">
            <collision name="collision">
               <geometry>
                  <plane>
                     <normal>0 0 1</normal>
                     <size>100 100</size>
                  </plane>
               </geometry>
            </collision>
            <visual name="visual">
               <geometry>
                  <plane>
                     <normal>0 0 1</normal>
                     <size>100 100</size>
                  </plane>
               </geometry>
               <material>
                  <ambient>0.8 0.8 0.8 1</ambient>
                  <diffuse>0.8 0.8 0.8 1</diffuse>
                  <specular>0.8 0.8 0.8 1</specular>
               </material>
            </visual>
         </link>
      </model>
      
      <model name="connected_system">
         <link name="p1">
            <pose>0.0 0.0 1.5 0 0 0</pose>
            <velocity_decay>
               <linear>100000.01</linear>
               <angular>100000.01</angular>
            </velocity_decay>
            <inertial>
               <inertia>
                  <ixx>600000</ixx>
                  <ixy>0.0</ixy>
                  <ixz>0.0</ixz>
                  <iyy>600000</iyy>
                  <iyz>0.0</iyz>
                  <izz>50000</izz>
               </inertia>
               <mass>1000000.0</mass>
            </inertial>
            <collision name="collision">
               <geometry>
                  <cylinder>
                     <length>3.0</length>
                     <radius>0.1</radius>
                  </cylinder>
               </geometry>
            </collision>
            <visual name="visual">
               <geometry>
                  <cylinder>
                     <length>3.0</length>
                     <radius>0.1</radius>
                  </cylinder>
               </geometry>
               <material>
                  <ambient>1 0 0 1</ambient>
                  <diffuse>1 0 0 1</diffuse>
               </material>
            </visual>
         </link>
         <link name="link1">
            <pose>0.0 -0.1 3.0 1.57 0 0</pose>
            <velocity_decay>
               <linear>2.1</linear>
               <angular>2.1</angular>
            </velocity_decay>
            <inertial>
               <inertia>
                  <ixx>0.3358e-03</ixx>
                  <ixy>0.0</ixy>
                  <ixz>0.0</ixz>
                  <iyy>0.3358e-03</iyy>
                  <iyz>0.0</iyz>
                  <izz>5.0e-06</izz>
               </inertia>
               <mass>0.001</mass>
            </inertial>
            <collision name="collision">
               <geometry>
                  <cylinder>
                     <length>0.2</length>
                     <radius>0.01</radius>
                  </cylinder>
               </geometry>
            </collision>
            <visual name="visual">
               <geometry>
                  <cylinder>
                     <length>0.2</length>
                     <radius>0.01</radius>
                  </cylinder>
               </geometry>
               <material>
                  <script>
                     <uri>file://media/materials/scripts/gazebo.material</uri>
                     <name>Gazebo/Black</name>
                  </script>
               </material>
            </visual>
         </link>
         <joint type="ball" name="hinge_1">
            <pose>0 0 1.5 0 0 0</pose>
            <parent>link1</parent>
            <child>p1</child>
            <axis>
               <use_parent_model_frame>true</use_parent_model_frame>
               <xyz>1 1 1</xyz>
               <dynamics>
                  <damping>10.04</damping>
               </dynamics>
            </axis>
         </joint>
      </model>
   </world>
</sdf>
`

iche033, 2024-05-08:
I think the `<velocity_decay>` param is not be supported in gz yet? I don't see it in sdf's [Link DOM](https://github.com/gazebosim/sdformat/blob/sdf14/src/Link.cc) or any references to it in gz-physics. 

Gazebo-classic reads the SDF element and applies linear / angular damping to the links. We would just need to do the same here.

iche033, 2024-05-14:
Do you need to use a ball joint? If you can use a revolute joint (or universal joint), the joint `<damping>` param should have an effect and make the free link come to a rest:

```xml
         <joint type="revolute" name="hinge_1">
            <pose>0 0 1.5 0 0 0</pose>
            <parent>link1</parent>
            <child>p1</child>
            <axis>
               <use_parent_model_frame>true</use_parent_model_frame>
               <xyz>1 0 0</xyz>
               <dynamics>
                  <damping>0.0001</damping>
               </dynamics>
            </axis>
         </joint>
```

vbalaji21, 2024-05-15:
@iche033 I tried the universal joint, it did not work and also saw the source code in gz-physics it did not support it. The revolute joint works for damping but my requirement is to have 3 DOF rotational joint so I need a ball joint. I tried a work around suggested by @azeey by having 2 dummy links to have it as 3 DOF rotational joint between the links. The movement is not the way I intended it to and shows a different motion. Since there is no damping in the links or joints, this is becoming a constraint and affecting my project. Is it possible to bring the changes done in Gazebo classic here ? It would be very helpful for my project

azeey, 2024-07-15:
We are not planning to work on this in the near term, but any help from the wider community would be appreciated.

vbalaji21, 2024-07-16:
@azeey I can work on it, if i get some top level guidance in doing it.