gazebosim/gz-physics #697: diff drive robot cannot rotate in bullet-feathestone

jsupratman13, 2024-10-25:
## Environment
* OS Version: Ubuntu 24.04
* Binary, `gz-physics7` installed along with ros2 jazzy
  
  

## Description
I apologize if this was address somewhere before. I'm working on a robot with differential drive using bullet-featherstone physics engine. The robot can move forward and backward without any issues but struggles to rotate in place. Instead of rotating, the robot slips, with no significant rotation. Even with the sample diff_drive.sdf, the vehicle is unable to turn or rotate in place. I also tried setting the caster friction to 0 but to no avail.

* Expected behavior: Diff drive robot can move can rotate in place
* Actual behavior: Diff drive robot does not rotate in place

## Steps to reproduce

1. `gz sim diff_drive.sdf --physics-engine gz-physics-bullet-featherstone-plugin`
2. `gz topic -t /model/vehicle_green/cmd_vel -m gz.msgs.Twist -p "linear: {x: 0.0}, angular: {z: 5.0}"`

## Output

https://github.com/user-attachments/assets/a12581db-0aff-4f01-a4e9-40b51799cfe2

iche033, 2024-10-29:
I think the default torsional friction value in bullet is causing this issue. As a workaround, try setting it to 0 to all the wheels and the ground plane, i.e. add this to the `<collision>` element in sdf:

```
          <surface>
            <friction>
              <torsional>
                <coefficient>0.0</coefficient>
              </torsional>
            </friction>
          </surface>

```

jsupratman13, 2024-10-30:
I see, thank you for your help!
Setting the wheel and the ground plane torsional friction to 0 works.

closed 2024-10-30 by jsupratman13

blue-assassin, 2025-04-10:
hello i am using gz harmonic 8.9.0 and opted for using bullet-featherstone for an issue revolving DART and prismatic joints in mobile robots . after encountering the problem mentioned in this issue i've tried the recommended solution unfortunately the problem still persists. Any suggestions would be very helpful please.