gazebosim/gz-physics #698: In Gazebo Harmonic, it occurs strange behavior

atinfinity, 2024-10-27:
## Environment
* OS Version:  Ubuntu 24.04 
* Source or binary build?: binary build
  
  

<details><summary>Result of <code>ros2 doctor --report</code></summary>

```bash
   PACKAGE VERSIONS
megarover_samples_ros2                    : latest=N/A, local=0.0.0
lifecycle                                 : latest=0.33.5, local=0.33.5
desktop                                   : latest=0.11.0, local=0.11.0
ament_cmake_python                        : latest=2.5.2, local=2.5.2
rosidl_typesupport_introspection_c        : latest=4.6.4, local=4.6.4
ament_cmake_pep257                        : latest=0.17.1, local=0.17.1
ros2action                                : latest=0.32.1, local=0.32.1
rosidl_adapter                            : latest=4.6.4, local=4.6.4
rqt_bag                                   : latest=1.5.4, local=1.5.4
rosbag2_storage_mcap                      : latest=0.26.5, local=0.26.5
rpyutils                                  : latest=0.4.1, local=0.4.1
qt_gui_py_common                          : latest=2.7.4, local=2.7.4
tf2                                       : latest=0.36.4, local=0.36.4
teleop_twist_keyboard                     : latest=2.4.0, local=2.4.0
rcl_logging_interface                     : latest=3.1.0, local=3.1.0
examples_rclcpp_minimal_service           : latest=0.19.4, local=0.19.4
python_cmake_module                       : latest=0.11.1, local=0.11.1
urdf                                      : latest=2.10.0, local=2.10.0
rosidl_typesupport_c                      : latest=3.2.2, local=3.2.2
quality_of_service_demo_py                : latest=0.33.5, local=0.33.5
tf2_sensor_msgs                           : latest=0.36.4, local=0.36.4
example_interfaces                        : latest=0.12.0, local=0.12.0
ament_cmake_target_dependencies           : latest=2.5.2, local=2.5.2
rqt_py_common                             : latest=1.6.0, local=1.6.0
launch_yaml                               : latest=3.4.2, local=3.4.2
shared_queues_vendor                      : latest=0.26.5, local=0.26.5
composition_interfaces                    : latest=2.0.2, local=2.0.2
turtlesim                                 : latest=1.8.3, local=1.8.3
ament_cmake_export_dependencies           : latest=2.5.2, local=2.5.2
rmw_dds_common                            : latest=3.1.0, local=3.1.0
dummy_sensors                             : latest=0.33.5, local=0.33.5
pendulum_control                          : latest=0.33.5, local=0.33.5
ament_cmake_export_targets                : latest=2.5.2, local=2.5.2
tf2_eigen                                 : latest=0.36.4, local=0.36.4
rcutils                                   : latest=6.7.2, local=6.7.2
pcl_msgs                                  : latest=1.0.0, local=1.0.0
ament_index_cpp                           : latest=1.8.1, local=1.8.1
demo_nodes_py                             : latest=0.33.5, local=0.33.5
rviz_common                               : latest=14.1.5, local=14.1.5
ament_cmake_cpplint                       : latest=0.17.1, local=0.17.1
topic_monitor                             : latest=0.33.5, local=0.33.5
ament_cmake_lint_cmake                    : latest=0.17.1, local=0.17.1
ament_cmake_export_interfaces             : latest=2.5.2, local=2.5.2
rttest                                    : latest=0.17.0, local=0.17.0
diagnostic_msgs                           : latest=5.3.5, local=5.3.5
rosidl_generator_c                        : latest=4.6.4, local=4.6.4
ros_environment                           : latest=4.2.1, local=4.2.1
statistics_msgs                           : latest=2.0.2, local=2.0.2
demo_nodes_cpp                            : latest=0.33.5, local=0.33.5
ros_base                                  : latest=0.11.0, local=0.11.0
std_msgs                                  : latest=5.3.5, local=5.3.5
ament_uncrustify                          : latest=0.17.1, local=0.17.1
rqt_gui_cpp                               : latest=1.6.0, local=1.6.0
ament_lint_auto                           : latest=0.17.1, local=0.17.1
image_tools                               : latest=0.33.5, local=0.33.5
ament_cmake_copyright                     : latest=0.17.1, local=0.17.1
rosidl_core_generators                    : latest=0.2.0, local=0.2.0
rosidl_parser                             : latest=4.6.4, local=4.6.4
intra_process_demo                        : latest=0.33.5, local=0.33.5
rosbag2_cpp                               : latest=0.26.5, local=0.26.5
[... 407 lines cut ...]
ros2 launch megarover_samples_ros2 vmegarover_with_sample_world.launch.py
ros2 run teleop_twist_keyboard teleop_twist_keyboard

type l key
type k key
```

## Supplemental information

On the other hands, there is no problem in **ROS 2 Humble** with **Gazebo Classic**.

```bash
mkdir -p ~/dev_ws/src
cd ~/dev_ws/src
git clone https://github.com/atinfinity/megarover_samples_ros2.git -b humble
cd ..
rosdep install -y -i --from-paths src/megarover_samples_ros2
colcon build --symlink-install
source ~/dev_ws/install/setup.bash
ros2 launch megarover_samples_ros2 vmegarover_with_sample_world.launch.py
ros2 launch megarover_samples_ros2 vmegarover_mapping.launch.py
ros2 run teleop_twist_keyboard teleop_twist_keyboard

type l key
```

azeey, 2024-12-16:
It looks like you have multiple sources of input for the robot publisher. From a quick look at your project, I see you have `joint_state_publisher`. Are you sure you need that? Can you check if there are multiple publishers to `/joint_states`?

atinfinity, 2024-12-17:
@azeey Thank you for your reply.

> Can you check if there are multiple publishers to /joint_states?

I checked it.

```bash
$ ros2 launch megarover_samples_ros2 vmegarover_with_sample_world.launch.py
```

```bash
$ ros2 topic info /joint_states -v
Type: sensor_msgs/msg/JointState

Publisher count: 2

Node name: base_topic_bridge
Node namespace: /
Topic type: sensor_msgs/msg/JointState
Topic type hash: RIHS01_a13ee3a330e346c9d87b5aa18d24e11690752bd33a0350f11c5882bc9179260e
Endpoint type: PUBLISHER
GID: 01.10.78.b8.94.49.71.06.2e.5d.b9.ef.00.00.1a.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (10)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite

Node name: joint_state_publisher
Node namespace: /
Topic type: sensor_msgs/msg/JointState
Topic type hash: RIHS01_a13ee3a330e346c9d87b5aa18d24e11690752bd33a0350f11c5882bc9179260e
Endpoint type: PUBLISHER
GID: 01.10.93.65.33.ea.ef.e4.fb.19.73.bc.00.00.18.03
QoS profile:
  Reliability: RELIABLE
  History (Depth): KEEP_LAST (10)
  Durability: VOLATILE
  Lifespan: Infinite
  Deadline: Infinite
  Liveliness: AUTOMATIC
  Liveliness lease duration: Infinite
```

So, I stopped to launch unnecessary "joint_state_publisher" node.
But, this problem was not resolved.