gazebosim/gz-physics #712: bullet-featherstone: support joint damping and friction parameters

scpeters, 2025-01-13:
Part of https://github.com/gazebosim/gz-physics/issues/545.

## Desired behavior

The [//joint/axis*/dynamics/damping](http://sdformat.org/spec?ver=1.11&elem=joint#dynamics_damping) and [//joint/axis*/dynamics/friction](http://sdformat.org/spec?ver=1.11&elem=joint#dynamics_friction) parameters are [read from SDFormat by the bullet-featherstone SDFFeatures::ConstructSdfModelImpl method](https://github.com/gazebosim/gz-physics/blob/gz-physics8/bullet-featherstone/src/SDFFeatures.cc#L788-L791) and written to the `m_jointDamping` and `m_jointFriction` member variables of [btMultibodyLink](https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBodyLink.h), but a [todo code comment in the btMultibodyLink.h](https://github.com/bulletphysics/bullet3/blob/master/src/BulletDynamics/Featherstone/btMultiBodyLink.h#L147-L148) indicates that those variables are not yet used.

We either need to implement support for these parameters in upstream bullet or implement an explicit joint damping workaround (see [gazebo::physics::ODEJoint::ApplyExplicitStiffnessDamping](https://github.com/gazebosim/gazebo-classic/blob/b22c6e15e52299865b31093b8feebc9ca19e26e8/gazebo/physics/ode/ODEJoint.cc#L1192-L1210) from gazebo-classic for reference).

## Alternatives considered

## Implementation suggestion

## Additional context

Sukhvansh2004, 2025-01-18:
Hi, can I work on this task?

iche033, 2025-01-21:
I think we'll need to first confirm that built-in support for joint damping / friction is  available in [bullet-featherstone implementation upstream](https://github.com/bulletphysics/bullet3/tree/master/src/BulletDynamics/Featherstone) or not. If not, I found this old post for doing it using [btMultibodyJointMotor](https://pybullet.org/Bullet/phpBB3/viewtopic.php?t=10942). Note that in our ode, we currently only [create a joint motor](https://github.com/gazebosim/gz-physics/blob/6b41b750f60d606a422a1e41c6e408fe37f42aa4/bullet-featherstone/src/JointFeatures.cc#L440-L454) when users set joint velocities.