gazebosim/gz-physics #737: CMake Error IMPORTED_LOCATION not set for imported target "octomap" configuration "Debug"

manan-peppermint, 2025-04-23:
I am trying to build gazebo fortress from source in ubuntu 22.04. While building packages with this command `colcon build --cmake-args ' -DBUILD_TESTING=OFF' ' -DCMAKE_BUILD_TYPE=Debug' --merge-install` I am getting this error `**CMake Error in CMakeLists.txt:  IMPORTED_LOCATION not set for imported target "octomap" configuration  "Debug". **`

I have installed `liboctomap-dev:amd64  1.9.7+dfsg-3` and `liboctomap1.9:amd64  1.9.7+dfsg-3`

[image]

Do i need to install any additional octomap libraries or any other libraries?

traversaro, 2025-04-23:
Which CMake version are you using?

scpeters, 2025-05-02:
are you building all of fortress from source?

manan-peppermint, 2025-05-02:
yes

scpeters, 2025-05-02:
I just started a colcon CI build of fortress on jammy

* [image] https://build.osrfoundation.org/job/ci__colcon_any-manual_ubuntu_jammy_amd64/7/

scpeters, 2025-05-02:
> I just started a colcon CI build of fortress on jammy
> 
> * [image] https://build.osrfoundation.org/job/ci__colcon_any-manual_ubuntu_jammy_amd64/7/

it's not resolving the dependencies properly...