gazebosim/gz-physics #766: change gravity at runtime

muttistefano, 2025-08-07:
Hello All,
I am trying to change gravity in runtime by using : 

`gz service -s /world/world_demo/set_physics   --reqtype gz.msgs.Physics   --reptype gz.msgs.Boolean   --timeout 2000   --req 'gravity { x: 0 y: 0 z: -2.8 }'`

which returns : 
`data: true`

The problem is that in the simulation, the gravity does not change.
I tried multiple physics plugins, and none of them allow this change.

I don't know if it's related, but I get this output on the gz terminal when I make the first call to the gz service to change gravity : 

> [gazebo-2] [Wrn] [Component.hh:144] Trying to serialize component with data type [N2gz4msgs7PhysicsE], which doesn't have `operator<<`. Component will not be serialized.
> [gazebo-2] [GUI] [Wrn] [Component.hh:189] Trying to deserialize component with data type [N2gz4msgs7PhysicsE], which doesn't have `operator>>`. Component will not be deserialized.

Am I doing something wrong, or is this simply not implemented?
Thanks.

iche033, 2025-08-07:
looking at the code, I see that `set_physics` eventually just [sets step size and rtf](https://github.com/gazebosim/gz-sim/blob/ec0cac5675c0a0ec4c744bc603f36ff4da40d6cb/src/SimulationRunner.cc#L396). I also noticed that [gravity is only set](https://github.com/gazebosim/gz-sim/blob/ec0cac5675c0a0ec4c744bc603f36ff4da40d6cb/src/systems/physics/Physics.cc#L1055) when the world is first created. So looks like changing gravity through `set_physics` is not supported yet.

azeey, 2025-12-29:
Fixed by #3189

closed 2025-12-29 by azeey